bluekitchen/btstack #520: Whitelist add/remove sync problem

amorniroli, 2023-08-10:
**Describe the bug**

In some circumstances, devices are not added to whitelist because they are recognized as "ON_CONTROLLER", even if they have been physically removed from the controller.

Besides, I've noticed that this snippet of code (hci.c line 6369 of v1.5.6.2-extra) is never called.

            if ((entry->state & LE_WHITELIST_ON_CONTROLLER) == 0){
				btstack_linked_list_remove(&hci_stack->le_whitelist, (btstack_linked_item_t *) entry);
				btstack_memory_whitelist_entry_free(entry);
            }

**To Reproduce**

Steps to reproduce the behavior:
1. Run example '...'
2. Connect to/from remove device '...'
3. Start action '...'

**Expected behavior**

A clear and concise description of what you expected to happen.

**HCI Packet Logs**
For all Bluetooth issues, please provide a full packet log that includes initial pairing. You need to zip the packet log as GitHub does not know how to handle .pklg files. 

How to get packet log depends on the port:
- For desktop ports, full packet logging is enabled by default and the packet log is available as `/tmp/hci_dump.pklg` (Mac/Linux), or `hci_dump.pklg' (Windows).
- For embedded ports, you can enable #define ENABLE_LOG_INFO in btstack_config.h and uncomment the call to 'hci_dump_init(..)' in the main() function to enable logging over UART Console or Segger RTT. Please capture text output and process using `btstack/tool/create_packet_log.py`. The tool will generate the text file into a .pklg file which can be analysed with Wireshark or Apple's PacketLogger tool.
- For ESP32: BTstack's `port/esp32/integrate.py` script creates one project per example. To enable packet logging, you need to edit app_main in main/main.c of the example project.

**Environment:  (please complete the following information):**
 - Current BTstack branch: [e.g. develop]
 - Bluetooth Controller [e.g. CSR 8510, TI CC2564C, ...]
 - Remote device: [e.g. iPhone X with iOS 13.3]

**Additional context**
Add any other context about the problem here.

mringwal, 2023-08-10:
Hi Alessandro. Could you try with latest version on master or develop? I think we've fixed this recently.

amorniroli, 2023-08-10:
> Hi Alessandro. Could you try with latest version on master or develop? I think we've fixed this recently.

Hi @mringwal , I confirm that on develop branch the bug is resolved. Sorry for bothering you: I always forget to check 'develop' branch prior to open an issue. 

Thank you very much!

closed 2023-08-10 by amorniroli

mringwal, 2023-08-10:
No problem. I'm sorry you had to run into that one - and it's not even on the master branch yet.