bluekitchen/btstack #521: Can't find a way to listen for HID reports in BLE device application

mohamed-AARAB, 2023-08-14:
I seem to not find an event (to use in the packet handler) related to receiving an HID report from the host as BLE HID device.

I can send the HID reports to the host. But I'm also trying to receive the HID reports from the host to then send them over USB to the device, which is a keyboard.

It seems counterintuitive if there is another way of doing it other than an event in the packet handler.

mohamed-AARAB, 2023-08-14:
Thanks for the reply.

Answering your question, I started with HID over Bluetooth classic, but I switched afterwards to BLE (HID over GATT) to benefit from the power saving aspect of it.

I wonder if there is a low level way of receiving those HID reports.

mringwal, 2023-08-15:
Do you have any numbers on energy for LE vs. Classic? Using sniff subrating, classic shouldn't be that bad, my old Apple Keyboard did run for months on 2 x  AA batteries.

If you really want to, you can hack the hids_device.c implementation and just forward the Set Report there somehow. We'll add it eventually.

mohamed-AARAB, 2023-08-16:
I don'

mohamed-AARAB, 2023-08-16:
I don't have numbers on the energy consumption between Bluetooth LE and classic.

But I think I will go back to using Bluetooth classic using sniff subrating as you said, keeping an eye on the BLE implementation on your library.

mohamed-AARAB, 2023-08-16:
Thanks a lot, I really appreciate your help.

closed 2023-08-16 by mohamed-AARAB