bluekitchen/btstack #527: avrcp_controller_get_element_attributes() corrupt avrcp_connection_t data.

sirius506, 2023-09-08:
**Describe the bug**

`avrcp_controller_get_element_attributes()` can be used to return value of specified attribute(s). If one attribute is specified, return packet size would be

packet_size = identifier(8) + NumAttributes(1) + AttributeID1 (4) = 13 octets

`connection->data` is used for the packet buffer, and is initialized as `connection->message_body` in `avrcp_controller_vendor_dependent_command_data_init()`.
However, `message_body[] `has only 11 octets length (AVRCP_MAX_COMMAND_PARAMETER_LENGTH).

**To Reproduce**

Steps to reproduce the behavior:

1. Eidt 'a2dp_sink_demo' example 'g' operation to return specified attribute.
2. Run example '.a2dp_sink_demo'
3. Press 'g' key to get attribute value.

**Expected behavior**

BTstack hangs.

**Environment:  (please complete the following information):**
 - Current BTstack branch: [master]
 - Target MCU [STM32H7B3I]
 - Bluetooth Controller [USB dongle]
 - Remote device: [Xiaomi phone Android 12]

sirius506, 2023-09-08:
Oops, made a typo. I meant press 'j' key to get the attribute.

mringwal, 2023-09-08:
Thanks for reporting. While we could just increase the buffer by 2 bytes, the same issue will appear once someone request 2 attributes. In this case, we'll try to collect the list of attributes into a bitmask and only expand it when sending. I'll update this issue when it's done.

commit 8b631c13 (2023-09-13): avrcp_controller: build get element attributes command on the fly, fix #527