bluekitchen/btstack #529: Bluetooth Characteristic Callback Issue

jhughesbiot, 2023-09-29:
Originally reported this bug to the pi pico arduino library here: https://github.com/earlephilhower/arduino-pico/issues/1735  but was directed to check here for a potential issue/fix as the implementation is based on btstack and the issue could be here.  Here is the full content of that issue:

Hardware: Pi Pico W
Library Version: 3.6.0 (earlephilhower/arduino-pico)

I am not sure if it I am doing something wrong with the Bluetooth implementation or this is an actual bug but it does appear the callbacks are not working as expected in the current Bluetooth stack.

I am connecting to a Nano33BLESense that has 1 characteristic that is setup to notify and writes a new value every 5 seconds.  I have tested the notify options with an android app to verify notification updates are working on the nano 

I modified the LECentral code slightly to try to get updates from characteristic subscriptions but these callbacks do not appear to work.   

Basic state order:
1. Connect to peripheral (Nano33BLESense)
2. Discover characteristics
3. Subscribe to characteristic
4. Wait for  characteristic updates

1-2 work as expected
3. Subscribe callback will always return BLEState of BLE_STATUS_OTHER_ERROR when you attempt to subscribe, no matter how many times you attempt.  If you perform a read of the characteristic, then subscribe, you will receive a subscribe call back status of BLE_STATUS_OK
4. Never receives any callback even after a subscribe callback of BLE_STATUS_OK

Additional testing on just reading the characteristic value every second resulted in a subscribe callback every time a read callback was completed.  No further subscriptions are occurring past the very first successful subscription but a successful read callback in turn receives a subscription callback. 

The gattCharacteristicNotification is never called despite successful subscription callback status and continual update of the characteristic in the nano.

Console Output showing connection and initial subscription failure, read, then success:
```
10:52:49.616 -> BLE ShieldService V2 found!
10:52:49.616 -> 
10:52:49.804 -> Device connected!
10:52:50.692 -> Service Discovered: :1800
10:52:50.692 -> Service Discovered: :1801
10:52:50.786 -> Service Discovered: :B8E06067-62AD-41BA-9231-206AE80AB550
10:52:50.786 -> Our service located!
10:52:50.879 -> Service discovery finished
10:52:50.972 -> Characteristic Discovered: F897177B-AEE8-4767-8ECC-CC694FD5FCEE, handle 0xC
10:52:50.972 -> Characteristic found: RX
10:52:50.972 -> Characteristic Discovered: BF45E40A-DE2A-4BC8-BBA0-E5D6065F1B4B, handle 0xF
10:52:50.972 -> Characteristic found: TX
10:52:50.972 -> Characteristic Discovered: 2FBC0F31-726A-4014-B9FE-C8BE0652E982, handle 0x11
10:52:50.972 -> Characteristic found: Baudrate
10:52:50.972 -> Characteristic Discovered: 1D132F28-5D85-11EE-8C99-0242AC120002, handle 0x13
10:52:50.972 -> Characteristic found: Intent
10:52:50.972 -> Characteristic discovery finished, status 1
10:52:50.972 -> Subscribing to charRX
10:52:50.972 -> Subscribing
10:52:51.066 -> 
10:52:51.066 -> Subscribe Callback: 4
10:52:51.066 -> BLE_STATUS_OTHER_ERROR
10:52:51.160 -> Read callback
10:52:51.160 -> Read Value:108 
10:52:51.160 -> 
10:52:51.160 -> Subscribe Callback: 0
10:52:51.160 -> BLE_STATUS_OK
```

One second reads showing a updates from nano but also a seemingly random subscription callback during every read and no gattCharacteristicNotification ever received:
```
`10:52:58.091 -> Subscribe Callback: 0
10:52:58.091 -> BLE_STATUS_OK
10:52:58.978 -> Reading...
10:52:59.071 -> Read callback
10:52:59.071 -> Read Value:80 
10:52:59.071 -> 
10:52:59.071 -> Subscribe Callback: 0
10:52:59.071 -> BLE_STATUS_OK
10:52:59.956 -> Reading...
10:53:00.049 -> Read callback
10:53:00.049 -> Read Value:80 
10:53:00.049 -> 
[... 432 lines cut ...]
  Serial.println("- Discovering central device...");
  delay(500);

  if (central) {
    Serial.println("* Connected to central device!");
    Serial.print("* Device MAC address: ");
    Serial.println(central.address());
    Serial.println(" ");
    lastWrite = millis();
    byte val = 0x50;
    while (central.connected()) {
      BLE.poll();
      central.poll();
        if( (millis() - lastWrite) > 5000){ 
          Serial.print("Writing RX: ");
          Serial.println(val);
          picoVoiceCharacteristicRx.writeValue(val);
          lastWrite = millis();
          val+=1;
        }
    }
    Serial.println("* Disconnected to central device!");
  }   
}
```

mringwal, 2023-10-04:
I didn't know anybody is still using this Arduino abstraction. If possible please use the native BTstack API.

Anyway, please check if you get an error for the call to device->subscribeForNotifications()

Also, please enable the packet logger to see if the Bluetooth communication is correct. See here: https://github.com/raspberrypi/pico-examples/pull/427/files

jhughesbiot, 2023-10-04:
What is the native BTstack API?  I'm not aware of any other arduino library for the pi pico w that enables bluetooth so this appears to be the only option currently out there.  

Since I'm not using this directly but using the project that is using this as it's core BT API, I don't really have a way to enable debugging directly as it doesn't appear to have the same options in the make file for that project.

mringwal, 2023-10-04:
Well, I was referring to use the Pico SDK that includes BTstack.

PacketLogger: there are enablePacketLogger() and enableDebugLogger() in the BTstackManager class.

Anyway, please also check if you get an error for the call to device->subscribeForNotifications()

jhughesbiot, 2023-10-05:
I'll have to spend some time with the pico SDK, haven't dealt with that as I've relied on this arduino library for past projects so it'll be some work to try to move everything to that dev style.

`Anyway, please also check if you get an error for the call to device->subscribeForNotifications()`

Are you referring to using this library as is?  Might have been lost in the original issue report as it was long but this always returns an error until you attempt to read a characteristic.  I can attempt to subscribe to a characteristic in a loop forever and it will always return BLE_STATUS_OTHER_ERROR.  Once you read the characteristic, that's where you start seeing subscribe callbacks (even though subscribes aren't happening) and never a characteristic notification callback.

earlephilhower, 2023-10-05:
Make sure you've enabled `Tools->Debug Port->Serial` or, since the `BTstack` library uses `printf()` to dump debugging, you won't be able to capture anything.

mringwal, 2023-10-06:
Thanks for the log. If possible, please remove the time stamps from your serial terminal, the conversion tool does not expect them as the HCI log already has time stamps.

--

After removing the PC timestamps, you can run the log through btstack/tool/create_packet_log.py to get a .pklg that can be analyzed in Wireshark. 

Here it is (zipped as GitHub doesnn't accept .pklg files): [arduino.pklg.zip](https://github.com/bluekitchen/btstack/files/12828756/arduino.pklg.zip)

--

The Arduino wrapper and your code seem to be ok. The stack first looks and finds the RX characteristic with ATT handles 0x000b - 0x000d:

	Characteristic Handle: 0x000B
		Properties: Read, Notify
		Value Handle: 0x000C
		UUID: F897177B-AEE8-4767-8ECC-CC694FD5FCEE

	Characteristic Handle: 0x000E
        ...

So we have:
  - 0x000b - characteristic declaration
  - 0x000c - characteristic value
  - 0x000d - client characteristic configuration descriptor (CCCD)

---

Next, the stack first tries to read the current value of the CCCD but gets an error!

Jan 01 11:00:13.453  ATT Send         Read By Type Request - Start Handle: 0x000C - End Handle: 0x000D - UUID: Client Characteristic Configuration  SEND  
	Read By Type Request - Start Handle: 0x000C - End Handle: 0x000D - UUID: Client Characteristic Configuration
	Opcode: 0x0008
	Starting Handle: 0x000C
	Ending Handle: 0x000D
	Attribute Type: 2902 (Client Characteristic Configuration)
Jan 01 11:00:13.453  L2CAP Send       Channel ID: 0x0004  Length: 0x0007 (07) [ 08 0C 00 0D 00 02 29 ]  SEND  
Jan 01 11:00:13.541  ATT Receive      Error Response - Attribute Handle: 0x000C - Error Code: Attribute Not Found (0x0A)  RECV  

If the remote side does have a CCCD at handle 0x000D, the ATT_READ_BY_TYPE_REQ should return it. However, it doesn't. 

Although you've tested with your Android phone, this looks like a bug in the Bluetooth Stack on the NanoBLE. Which one is this? 

It looks like BTstack uses different, but valid ATT requests to enable notifications, which are handled incorrectly by the Peripheral.

--

Could you try with a different Peripheral / different Bluetooth Stack to see if your Central on top of BTstack on Pico works correctly then?

Could you ask the developer of the stack on the NanoBLE about this behaviour and provide your log?

mringwal, 2023-10-06:
Looks like the NanoBLE33 is running this code here - didn't know that Arduino implemented an LE Host Stack:
https://github.com/arduino-libraries/ArduinoBLE/blob/dff9e414abc20df491b46d0b38b360eb1a3507fb/src/utility/ATT.cpp#L1105

Maybe the end condition in the loop is off by one or something like that.

Could you print:
- GATT.attributeCount()
- readByTypeReq->startHandle
- (readByTypeReq->endHandle
before entering the for loop (line 1124) and in the loop (line 1128)
- i
- handle
- attribute->type()
to see if it exits too early?

jhughesbiot, 2023-10-06:
I don't have another peripheral I can test at the moment.

Here is the output with that modification from the nano:

> AttrCount: 20
> StarHandle: 10
> EndHandle: 20
> i: 9
> Handle: 10
> Type: 10240
> 
> AttrCount: 20
> StarHandle: 10
> EndHandle: 20
> i: 10
> Handle: 11
> Type: 10243
> 
> AttrCount: 20
> StarHandle: 10
> EndHandle: 20
> i: 12
> Handle: 13
> Type: 10496
> 
> AttrCount: 20
> StarHandle: 10
> EndHandle: 20
> i: 13
> Handle: 14
> Type: 10243
> 
> AttrCount: 20
> StarHandle: 10
> EndHandle: 20
> i: 15
> Handle: 16
> Type: 10243
> 
> AttrCount: 20
> StarHandle: 10
> EndHandle: 20
> i: 17
> Handle: 18
> Type: 10243
> 
> AttrCount: 20
> StarHandle: 10
> EndHandle: 20
> i: 19
> Handle: 20
> Type: 10496
> 
> AttrCount: 20
> StarHandle: 20
> EndHandle: 20
> i: 19
> Handle: 20
> Type: 10496
> 
> AttrCount: 20
> StarHandle: 12
> EndHandle: 13
> i: 11
> Handle: 12
> Type: 10243
> 
> AttrCount: 20
> StarHandle: 12
> EndHandle: 13
> i: 12
> Handle: 13
> Type: 10496

mringwal, 2023-10-07:
I think I found it. The attribute->type() only returns BLETypeService (2800), BLETypeCharacteristic (2803) or BLETypeDescriptor (2900), but never 2902. With this

if (attribute->type() == readByTypeReq->uuid)

is never true for our query - although there's code to explicitly handle 2902 but it's not possible to reach it.

Could you try this hack/patch:

diff --git a/src/local/BLELocalDescriptor.cpp b/src/local/BLELocalDescriptor.cpp
index c466897..0c81fe7 100644
--- a/src/local/BLELocalDescriptor.cpp
+++ b/src/local/BLELocalDescriptor.cpp
@@ -40,6 +40,9 @@ BLELocalDescriptor::~BLELocalDescriptor()

 enum BLEAttributeType BLELocalDescriptor::type() const
 {
+  if (strcmp("2902", uuid.str()) == 0){
+    return (enum BLEAttributeType) 0x2902;
+  }
   return BLETypeDescriptor;
 }

This let's the CCCD return 0x2902 as type() and should enable the code to handle ATT Read By Type for UUID 2902.

mringwal, 2023-10-08:
Reported as https://github.com/arduino-libraries/ArduinoBLE/issues/326

jhughesbiot, 2023-10-10:
I updated the code, slightly modified so it would compile as follows:

```
enum BLEAttributeType BLELocalDescriptor::type() const
{
  if (strcmp("2902", uuid()) == 0){
    return (enum BLEAttributeType) 0x2902;
  }
  return BLETypeDescriptor;
}
```

No change in behavior.

jhughesbiot, 2023-10-10:
That is how the original code looked, just returned the type with no check.

mringwal, 2023-10-10:
Oh, the diff didn't survive. The idea was to have the BLELocalDescriptor return 0x2902 if the UUID is 2902, such that this comparison in the ATT.cpp code becomes true:

`if (attribute->type() == readByTypeReq->uuid)`

(as the readByTypeReq->uuid is 2902 and the BLELocalDescriptor should have UUID "2902" as well).

Again, a correct way (give the 'cross-layer optimization': ATT.cpp is implementing the logic for GATT) would be to do a switch on attribute->type() and for BLETypeDescriptor, compare the attribute->uuid() to the request UUID.

mringwal, 2023-10-10:
> GitHub munged his `diff`, but I think he meant to write:
> 
> ```
> enum BLEAttributeType BLELocalDescriptor::type() const
> {
>    return BLETypeDescriptor;
> }
> ```
> 
> I.e. just return what it has, no strcmp() or any other comparison needed.

Could you also log attribute->type() and paste the output?

jhughesbiot, 2023-10-11:
Where do you want me to add that output?  It is already in the first output with the updatedATT.cpp file and that hasn't changed from the previous output.

mringwal, 2023-10-11:
Btw, it would be convenient to print the attribute->type() as hex.
Could you also print attribute->uuid()?

Anyway, when the hack/patch works, it should list type 0x2902 = 10498.
If that's not the case, then it's not working. Please try to get it to work.

When it attribute->type() returns 0x2902, the ATT Read By Type Request should be answered correctly (at least that's what I read in the code)

jhughesbiot, 2023-10-11:
Changed the print statement in the ATT.cpp readByTypeReq for loop and get the output below.  I do see the expected 2902 value showing up but still no change in behavior.  Not sure if there's something farther down the line in the code still having issues.

> AttrCount: 20
> StarHandle: 10
> EndHandle: 20
> i: 9
> Handle: 10
> UUID: B8E06067-62AD-41BA-9231-206AE80AB550
> readByTypeReqUUID: 2803
> Type: 2800
> 
> AttrCount: 20
> StarHandle: 10
> EndHandle: 20
> i: 10
> Handle: 11
> UUID: F897177B-AEE8-4767-8ECC-CC694FD5FCEE
> readByTypeReqUUID: 2803
> Type: 2803
> 
> AttrCount: 20
> StarHandle: 10
> EndHandle: 20
> i: 12
> Handle: 13
> UUID: 2902
> readByTypeReqUUID: 2803
> Type: 2902
> 
> AttrCount: 20
> StarHandle: 10
> EndHandle: 20
> i: 13
> Handle: 14
> UUID: bf45e40a-de2a-4bc8-bba0-e5d6065f1b4b
> readByTypeReqUUID: 2803
> Type: 2803
> 
> AttrCount: 20
> StarHandle: 10
> EndHandle: 20
> i: 15
> Handle: 16
> UUID: 2fbc0f31-726a-4014-b9fe-c8be0652e982
> readByTypeReqUUID: 2803
> Type: 2803
> 
> AttrCount: 20
> StarHandle: 10
> EndHandle: 20
> i: 17
> Handle: 18
> UUID: 1d132f28-5d85-11ee-8c99-0242ac120002
> readByTypeReqUUID: 2803
> Type: 2803
> 
> AttrCount: 20
> StarHandle: 10
> EndHandle: 20
> i: 19
> Handle: 20
> UUID: 2902
> readByTypeReqUUID: 2803
> Type: 2902
> 
> AttrCount: 20
> StarHandle: 20
> EndHandle: 20
> i: 19
> Handle: 20
> UUID: 2902
> readByTypeReqUUID: 2803
> Type: 2902
> 
> AttrCount: 20
> StarHandle: 12
> EndHandle: 13
> i: 11
> Handle: 12
> UUID: F897177B-AEE8-4767-8ECC-CC694FD5FCEE
> readByTypeReqUUID: 2902
> Type: 2803
> 
> AttrCount: 20
> StarHandle: 12
> EndHandle: 13
> i: 12
> Handle: 13
> UUID: 2902
> readByTypeReqUUID: 2902
> Type: 2902

Generated from this updated code in the ATT.cpp readByTypeReq function.
```
    Serial.print("\nAttrCount: ");
    Serial.println(GATT.attributeCount());
    Serial.print("StarHandle: ");
    Serial.println(readByTypeReq->startHandle);
    Serial.print("EndHandle: ");
    Serial.println(readByTypeReq->endHandle);
    Serial.print("i: ");
    Serial.println(i);
    Serial.print("Handle: ");
    Serial.println(handle);
    Serial.print("UUID: ");
    Serial.println(attribute->uuid());
    Serial.print("readByTypeReqUUID: ");
    Serial.println(readByTypeReq->uuid, HEX);
    Serial.print("Type: ");
    Serial.println(attribute->type(),HEX);
```

earlephilhower, 2023-10-12:
I'm not sure if it would help here, but the latest `git` `master` now supports the built-in BTstack logging through the IDE `Debug` menus.

mringwal, 2023-10-13:
Ok, so the attribute->type is now fixed to 2902, but it's not working. Could you a) provide a new HCI trace from the Pico? b) add more debug output to see if the code after ` if (attribute->type() == 0x2902) {` is actually executed and what value is returned. For BTstack, I don't think the actual value is relevant, but the it will use the attribute handle to do a write request and enable notifications.

jhughesbiot, 2023-10-17:
@earlephilhower 
Are you referring to the lines:
```
 BTstack.enablePacketLogger();
 BTstack.enableDebugLogger();
```
Or is there some other logging I am able to turn on to help troubleshoot?

@mringwal 

**Code modified to print after if statement:**
```
} else if (attribute->type() == 0x2902) {
        Serial.println("\nAttribute 0x2902 found");
        BLELocalDescriptor* descriptor = (BLELocalDescriptor*)attribute;

        // add the handle
        memcpy(&response[responseLength], &handle, sizeof(handle));
        responseLength += sizeof(handle);

        // add the value
        int valueSize = min((uint16_t)(mtu - responseLength), (uint16_t)descriptor->valueSize());
        memcpy(&response[responseLength], descriptor->value(), valueSize);
        responseLength += valueSize;

        response[1] = 2 + valueSize;

        Serial.print("Response: ");
        Serial.println(response[1]);

        break; // all done
      }
```

 **Results in this output:**
> AttrCount: 20
> StarHandle: 10
> EndHandle: 20
> i: 9
> Handle: 10
> UUID: B8E06067-62AD-41BA-9231-206AE80AB550
> readByTypeReqUUID: 2803
> Type: 2800
> 
> AttrCount: 20
> StarHandle: 10
> EndHandle: 20
> i: 10
> Handle: 11
> UUID: F897177B-AEE8-4767-8ECC-CC694FD5FCEE
> readByTypeReqUUID: 2803
> Type: 2803
> 
> AttrCount: 20
> StarHandle: 10
> EndHandle: 20
> i: 12
> Handle: 13
> UUID: 2902
> readByTypeReqUUID: 2803
> Type: 2902
> 
> AttrCount: 20
> StarHandle: 10
> EndHandle: 20
> i: 13
> Handle: 14
> UUID: bf45e40a-de2a-4bc8-bba0-e5d6065f1b4b
> readByTypeReqUUID: 2803
> Type: 2803
[... 216 lines cut ...]
> Read Value:80 
> [01:04:03.531] EVT <= A0 03 40 00 00 
> 
> Subscribe Callback: 0
> BLE_STATUS_OK
> [01:04:03.531] CMD => 35 0C 05 01 40 00 01 00 
> [01:04:03.532] EVT <= 6E 00 
> [01:04:03.617] EVT <= 13 05 01 40 00 01 00 
> Reading...
> [01:04:08.421] LOG -- gatt_client.c.124: GATT client timeout start, handle 0x40
> [01:04:08.422] ACL => 40 00 07 00 03 00 04 00 0A 0C 00 
> [01:04:08.422] EVT <= 6E 00 
> [01:04:08.423] EVT <= 79 02 04 00 
> [01:04:08.570] ACL <= 40 20 06 00 02 00 04 00 0B 51 
> [01:04:08.570] LOG -- gatt_client.c.132: GATT client timeout stop, handle 0x40
> [01:04:08.570] EVT <= A5 07 40 00 0C 00 01 00 51 
> Read callback
> Read Value:81 
> [01:04:08.571] EVT <= A0 03 40 00 00 
> 
> Subscribe Callback: 0
> BLE_STATUS_OK
> [01:04:08.571] CMD => 35 0C 05 01 40 00 01 00 
> [01:04:08.572] EVT <= 6E 00 
> [01:04:08.741] EVT <= 13 05 01 40 00 01 00

mringwal, 2023-10-20:
Nice. Progress! Here's the packet log for Wireshark

[arduino.pklg.zip](https://github.com/bluekitchen/btstack/files/13050985/arduino.pklg.zip)

Jan 01 12:03:58.760  ATT Receive      Read By Type Response  RECV  
	Read By Type Response
	Opcode: 0x0009
	Length: 0x0004 (4)
	Characteristic Handle: 0x000D
		Raw Value: : 0D00 0000

With the changes, the ATT Read By Type Request is answered correctly: BTstack learns that the CCCD Handle is 0x000d.

Ok, next BTstack sends a write request to enable notifications.

Jan 01 12:03:58.761  ATT Send         Write Request - Handle:0x000D - F897177B-AEE8-4767-8ECC-CC694FD5FCEE - Value: 0100  SEND  
	Write Request - Handle:0x000D - F897177B-AEE8-4767-8ECC-CC694FD5FCEE - Value: 0100
	Opcode: 0x0012
	Attribute Handle: 0x000D (13)
	Value: 0100

Looks good. The Nano responds with:

Jan 01 12:03:58.850  ATT Receive      Error Response - Attribute Handle: 0x000D - Error Code: Write Not Permitted (0x03) - Write Not Permitted  RECV  
	Error Response - Attribute Handle: 0x000D - Error Code: Write Not Permitted (0x03)
	Opcode: 0x0001
	Request Opcode In Error: 0x0012 (Write Request)
	Attribute Handle In Error: 0x000D (13)
	Error Code: 0x0003 (Write Not Permitted)

Quick look into the code, it looks like the previous hack broke the write request.
Going forward, could you patch 

https://github.com/arduino-libraries/ArduinoBLE/blob/dff9e414abc20df491b46d0b38b360eb1a3507fb/src/utility/ATT.cpp#L1306

and replace BLETypeDescriptor with `((enum BLEAttributeType) 0x2902)`? It should then enter that scope again.