bluekitchen/btstack #531: BLE Notifications: att_server_handle_can_send_now should not be static, or the documentation is incorrect.

shabaz123, 2023-10-09:
**Describe the bug**
I'm using the Pico W as a BLE Peripheral, with custom characteristics. I'd like to send notifications from the Pico W.

From what I've read, I would need to call the **att_server_handle_can_send_now()** function to do that, so that it can generate an event that I can pick up and then call **att_server_notify**.

This is described on page 65/66 of the Bluekitchen user manual https://bluekitchen-gmbh.com/btstack.pdf

However, that function **att_server_handle_can_send_now()** is a static function, in **att_server.c**, and has no prototype in **att_server.h**.

This is either a documentation bug, or it is a code bug, because the documentation does not match the code.

**To Reproduce**

Steps to reproduce the behavior:
1. See my code at https://gist.github.com/shabaz123/a3fc517ac8ba5cf8bd3f776da1055ef3
2. On line 162 of my code, you can see the use of the function **att_server_handle_can_send_now()**
3. If the code is built, it will fail during linking, because that function is static.

**Expected behavior**
I would expect that function att_server_handle_can_send_now() to be not static, or I would expect the documentation to explain how to implement notifications.

In particular, it would be nice to see an example with custom characteristics and Notifications, using the Pi Pico, as part of any examples or documentation.

My current workaround (that works) is to not use att_server_handle_can_send_now() and instead, just directly call att_server_notify. But I don't know if this is recommended or not.

**Environment:  (please complete the following information):**
  - Current BT stack: The latest provided with the Pi Pico SDK from the raspberrypi.com website
 - Bluetooth controller: The one on the Pi Pico W, i.e. CYW43
 - Remote device: Android (Samsung Galaxy)

**Additional context**
You can see the discussion on the Raspberry Pi forum here: https://forums.raspberrypi.com/viewtopic.php?p=2142618#p2142618

mringwal, 2023-10-09:
Hi. Where does it say that you should call `att server request can send now`?

We have: "Similar to other protocols, it might be not possible to send any time. To send a Notification, you can call att_server request_can_send_now to receive a ATT EVENT CAN SEND NOW event."

Ah.. that should have been `att_server_request_can_send_now_event`, which we have deprecated.

Please call: `att_server_request_to_send_notification` to request a callback to your code when you can send.
Directly calling `att_server_notify` will return an error if it couldn't sent (due to outgoing buffer already used, ACL buffer in Bluetooth Controller full, ...)

We'll update the documentation.

shabaz123, 2023-10-10:
Thanks, 
I tried that, and it works.
Code snippet in case anyone needs it:

```
// called by the stack, to send a notification
STATIC void notify_callback(void *context) {
    printf("btstack_notify calling att_server_notify\n");
    int err = att_server_notify(((notify_t*)context)->conHandle ,
                                ATT_CHARACTERISTIC_0000CC02_0000_1000_8000_00805F9B34FB_01_VALUE_HANDLE,
                                (const uint8_t*)(((notify_t*)context)->data),
                                ((notify_t*)context)->len);
    if (err) {
        printf("btstack_notify - error!\n");
    }
}

// use ble_send_notification to send a notification
void
ble_send_notification(char* data)
{
    if (NotifyState) {
        printf("ble_send_notification\n");
        notify.data = data;
        notify.len = strlen(data);
        notify.conHandle = con_handle;
        context_registration.callback = &notify_callback;
        context_registration.context = &notify;
        // tell the stack to call the notify_callback if it is OK to send a notification:
        att_server_request_to_send_notification(&context_registration, con_handle);
    }
}
```

closed 2023-10-10 by shabaz123