bluekitchen/btstack #532: ESP32 issue in HFP with microphone

christianfrancouib, 2023-10-09:
Hi all, I'm using the ESP32 microcontroller for running the hfp_demo. I was struggling the first time (see https://github.com/bluekitchen/btstack/issues/486), but with your help, I could make it work. 

But now, after several tests, an issue is appearing every once in a while after some time (random reproduction).

*Describe the bug**

As I said, after some time the CPU shows a `core Dump issue` when I'm sometimes on a call. I have attached the logging when is reproduced, you can see the `dump` report at the end of the file.

[btstreaming_issue2_last_commit_346b3650caa2a.txt](https://github.com/bluekitchen/btstack/files/12848723/btstreaming_issue2_last_commit_346b3650caa2a.txt)

The assert reported by the BTStack library is this one:

```
assert failed:
btstack_audio_esp32_source_process_buffer mic.c:419 (bytes_read == data_len)
```
The BTStack version used is the `develop` branch in the commit `346b3650caa2a`.

mringwal, 2023-10-10:
Hi. The assert indicates a bug or misunderstanding of the ESP32 I2S driver interface. When we get the I2S_EVENT_RX_DONE event, the buffer should be full and the numbers read should match the expected samples.

We've implemented a new driver for ESP-iDF v5.x which has a new driver architecture. Could you try, e.g. with our regular examples if this got fixed in v5.x and if yes, update your project.

If you want/must stay at v4.x, you can try to patch it somehow. In the tx code, we already test if we couldn't write a single byte and then ignore this. You could try to just exit the function if the numbers of bytes are zero. This would mean, we got an event too many.

christianfrancouib, 2023-10-10:
Thank you, I'll have a look.

christianfrancouib, 2023-10-13:
Hi Mathias, and thank you again for your time and help.

Regarding the version, I want to stay at v4.x, so what I did was to return from the function when I see that `bytes_read != data_len`, and log to the console when this happens. Now the system does not block.

But after several tests, what I observe is that when I pair the device from my mobile phone (iPhone) every once in a while the issue occurs, producing slots in the audio. But, when I pair the device from my Windows 10 pc, this never happens (I have done recordings  > 30 minutes). Actually, you can see in the `SCO demo statistics` that in the phone case the number of `bad frames` es > 0, but for the Widows pc, it always gives me 0 (no `bad frames`).

Some clue I observed that may help is that when I pair from Windows, the event that fires is the `HFP_SUBEVENT_AUDIO_CONNECTION_ESTABLISHED`, but when it's paired from the phone the event that fires is the `HFP_SUBEVENT_CALL_ANSWERED`. Does this tell you something?

thank you again.

mringwal, 2023-10-13:
When you get bytes_read != data_len, is bytes_read = 0 or a different value?

SCO is not error free - as it's time limited an there is only up to 2 or so retries, so bad frames are expected. With the mSBC PLC it's not that bad.

If you only need audio from phone to ESP32, why don't you use A2DP instead? If you need, you can reduce the buffering (higher chance of underruns in exchange for lower latency).

christianfrancouib, 2023-10-16:
**_When you get bytes_read != data_len, is bytes_read = 0 or a different value?_**
bytes_read = 0 is the answer.

**_If you only need audio from phone to ESP32, why don't you use A2DP instead?_**
In the intended application, I want the phone to act as the master and the device as the slave, i.e. I need to search for the device from the phone (Master), pair it, and control when I start to transmit the audio. What I see in the A2DP implementation is that the device acts like the Master, pairing with the phone and sending the audio. Is there a way of doing it as I intend to?  

### **If you need, you can reduce the buffering (higher chance of underruns in exchange for lower latency).**
You mean in the actual hfp application? how can I do that?

christianfrancouib, 2023-10-18:
After several tests, I see that when I start recording from my W10 or W11 machine the link created is somehow different from when I connect from my phone. 

In the phone the logging prints `Call answered`, and at the end of statistics it shows some `good frames` and some `bad frames`. 
But in the case of the W10/W11 machine the `Call answered` is not printed and the statistics always show 0 for both `good frames` and `bad frames`. 

What is the difference between the two connections? and given that establishing the connection without the `Call event` works okay, is there a way of establishing it this way from the phone?... 

Thank you again for your time.

mringwal, 2023-10-18:
> **_When you get bytes_read != data_len, is bytes_read = 0 or a different value?_** bytes_read = 0 is the answer.
> 

Thanks. I'll add a quick return to the v4 code then.

> **_If you only need audio from phone to ESP32, why don't you use A2DP instead?_** In the intended application, I want the phone to act as the master and the device as the slave, i.e. I need to [...]

Sorry, that's not possible with a phone. Phones only implement the A2DP Source Role.
> 
> **If you need, you can reduce the buffering (higher chance of underruns in exchange for lower latency).** You mean in the actual hfp application? how can I do that?

No, that was related to A2DP which commonly uses a playback buffer of around 100 ms (which is fine for music, but not so great for voice communication).

mringwal, 2023-10-18:
> But in the case of the W10/W11 machine the `Call answered` is not printed and the statistics always show 0 for both `good frames` and `bad frames`.

If there are zero good frames and zero bad frames, either you didn't receive any audio, or the statistic is incorrect.

> 
> What is the difference between the two connections? and given that establishing the connection without the `Call event` works okay, is there a way of establishing it this way from the phone?...

A phone will signal the call status (e.g. idle, ringing, connected), while your PC probably only opens the HFP audio connection. The signaling should not have any impact on the audio itself as it's just a few RFCOMM packets with AT commands to synchronize the call state, but there's no additional communication once the call/audio has started.

christianfrancouib, 2023-10-19:
Do you know if from the phone we can only open the HFP audio connection as it does the PC?

mringwal, 2023-10-19:
HSP does not have call state signaling, so it might work for you.
If you just use the HFP HF device as microphone from an app, I don't think call state signaling will happen, either.

mringwal, 2023-10-19:
I don't think call state signaling will happen on a mobile phone for a regular app (that isn't Zoom, Signal, Teams, ..)

christianfrancouib, 2023-10-19:
But if I develop the app, is it available ?

mringwal, 2023-10-19:
I don't know if/how you can use call signaling on Android and/or iOS. I would assume that by default, there is no call state signaling.

christianfrancouib, 2023-10-23:
Hey Mithias, yes the change is ok. 

BTW I'm trying to run the `hsp` example with the sine wave sending.  But, when I couldn't really pair the device. It tries for some time and then disconnects by itself. 

This is what it shows:

[image]

mringwal, 2023-10-23:
Closing this. 

For HSP, please create a new issue and always attach an HCI packet log.

closed 2023-10-23 by mringwal