bluekitchen/btstack #539: HSP example not pairing

christianfrancouib, 2023-10-23:
**Describe the bug**

Hi, I'm trying to use the `hsp` example code with an ESP32 board using the sine wave mode, but I cant pair with the device. 

**Expected behavior**

A clear pairing.

**HCI Packet Logs**
Attached is the log in .txt file. In the log, I try to pair with it 3 times, but I couldnt do it.
[hsp_log_hci_2.txt](https://github.com/bluekitchen/btstack/files/13074510/hsp_log_hci_2.txt)

Thank you

mringwal, 2023-11-10:
Sorry, I've missed this. In your log, there's a link key request after the connection was established and your esp32 answers with "don't have a key". It's possible that the other side has a key from an earlier pairing. Please try to delete the device on the device you connect to the ESP32 and/or try another one. 

Feel free to post another packet log.

mringwal, 2023-11-27:
no response

closed 2023-11-27 by mringwal