bluekitchen/btstack #544: [regression]: HID parser fails to correctly parse Xbox Wireless HID descriptor

ricardoquesada, 2023-11-11:
I created a test case that fails on "develop" branch, but passes on "master" branch.

https://github.com/bluekitchen/btstack/pull/543

Possibly the regression was caused by this CL:
https://github.com/bluekitchen/btstack/commit/dfb01e777425a3bc67a37a54368ece74b1630f47
(although I haven't done the bisect)

ricardoquesada, 2023-11-12:
I confirm that the regression was introduced in https://github.com/bluekitchen/btstack/commit/dfb01e777425a3bc67a37a54368ece74b1630f47

mringwal, 2023-11-13:
Thanks for reporting. The regression was caused by the workaround for the TankMouse. I've fixed the condition to not cause problems to others. Fixed in 1a05cde

closed 2023-11-13 by mringwal

ricardoquesada, 2023-11-13:
Thanks for the quick fix!