bluekitchen/btstack #545: Compilation warning / error on latest "develop" branch + ESP-IDF 5.1

ricardoquesada, 2023-11-13:
Compile latest develop branch with ESP-IDF 5.1 and you will get this error messasge:

```
In file included from /builds/ricardoquesada/bluepad32/src/components/btstack/src/btstack_hid_parser.c:46:
/builds/ricardoquesada/bluepad32/src/components/btstack/src/btstack_hid_parser.c: In function 'hid_find_next_usage':
/builds/ricardoquesada/bluepad32/src/components/btstack/src/btstack_debug.h:109:70: error: format '%x' expects argument of type 'unsigned int', but argument 5 has type 'uint32_t' {aka 'long unsigned int'} [-Werror=format=]
  109 | #define HCI_DUMP_LOG(log_level, format, ...) hci_dump_log(log_level, "%s.%u: " format, BTSTACK_FILE__, __LINE__, ## __VA_ARGS__)
      |                                                                      ^~~~~~~~~
/builds/ricardoquesada/bluepad32/src/components/btstack/src/btstack_debug.h:113:33: note: in expansion of macro 'HCI_DUMP_LOG'
  113 | #define log_debug(format, ...)  HCI_DUMP_LOG(HCI_DUMP_LOG_LEVEL_DEBUG, format,  ## __VA_ARGS__)
      |                                 ^~~~~~~~~~~~
/builds/ricardoquesada/bluepad32/src/components/btstack/src/btstack_hid_parser.c:263:21: note: in expansion of macro 'log_debug'
  263 |                     log_debug("Usage Min - Usage Max [%04x..%04x] < Report Count %u", parser->usage_minimum, parser->usage_maximum, parser->required_usages);
      |                     ^~~~~~~~~
/builds/ricardoquesada/bluepad32/src/components/btstack/src/btstack_debug.h:109:70: error: format '%x' expects argument of type 'unsigned int', but argument 6 has type 'uint32_t' {aka 'long unsigned int'} [-Werror=format=]
  109 | #define HCI_DUMP_LOG(log_level, format, ...) hci_dump_log(log_level, "%s.%u: " format, BTSTACK_FILE__, __LINE__, ## __VA_ARGS__)
      |                                                                      ^~~~~~~~~
/builds/ricardoquesada/bluepad32/src/components/btstack/src/btstack_debug.h:113:33: note: in expansion of macro 'HCI_DUMP_LOG'
  113 | #define log_debug(format, ...)  HCI_DUMP_LOG(HCI_DUMP_LOG_LEVEL_DEBUG, format,  ## __VA_ARGS__)
      |                                 ^~~~~~~~~~~~
/builds/ricardoquesada/bluepad32/src/components/btstack/src/btstack_hid_parser.c:263:21: note: in expansion of macro 'log_debug'
  263 |                     log_debug("Usage Min - Usage Max [%04x..%04x] < Report Count %u", parser->usage_minimum, parser->usage_maximum, parser->required_usages);
      |                     ^~~~~~~~~
cc1: some warnings being treated as errors
[1013/1184] Building C object esp-idf/btstack/CMakeFiles/__idf_btstack.dir/src/btstack_audio.c.obj
[1014/1184] Building C object esp-idf/btstack/CMakeFiles/__idf_btstack.dir/src/mesh/mesh_upper_transport.c.obj
ninja: build stopped: subcommand failed.
HINT: The issue is better to resolve by replacing format specifiers to 'PRI'-family macros (include <inttypes.h> header file).
```

mringwal, 2023-11-13:
please check again d30de50

ricardoquesada, 2023-11-13:
thanks for the quick fix.

closed 2023-11-13 by ricardoquesada