bluekitchen/btstack #548: HCI transport: Add support for Linux HCI socket?

AminoACID123, 2023-11-23:
**Is your feature request related to a problem? Please describe.**
For some reason I need to run Btstack on Linux with a local  controller attached and directly accessible via Linux HCI `socket`. Currently for posix port, Btstack supports uart and usb. I think it would be nice if Btstack supports transport over Linux HCI socket.

**Describe the solution you'd like**
Add support for transport over Linux HCI socket, similar to that in [zephyr](https://github.com/zephyrproject-rtos/zephyr/blob/eeb6bf7dd998c8c191f58df692097bfcfaf1e3bf/drivers/bluetooth/hci/userchan.c#L290) or [nimble](https://github.com/apache/mynewt-nimble/blob/619032f52a093ace90dc6809df224a4b73ad4a71/nimble/transport/socket/src/ble_hci_socket.c#L593)

mringwal, 2023-11-23:
We've recently added support for FreeBSD's Netgraph Bluetooth HCI driver on the develop branch https://github.com/bluekitchen/btstack/blob/develop/port/freebsd-netgraph/hci_transport_netgraph.c - which required to disable FreeBSD's L2CAP layer, though. 

Do you want/need to keep using BlueZ in parallel, or do you only require BlueZ to load/initialized the Bluetooth Controller (which isn't supported by BTstack directly yet)?

mringwal, 2023-11-23:
Good, running two Host Stacks with one Controller would be a disaster :)
BTstack should be fine once someone implemented hci_transport_t via BueZ' HCI socket. Given the references to Zephyr and NimBLE, it looks like it should just work out of the box (it didn't with FreeBSD which didn't want to send/receive ACL directly).

AminoACID123, 2024-04-22:
I am currently implementing this feature. However, I encountered some problems. I wrote a custom `hci_transport_t` driver which uses linux bluez socket to send/receive packets. The driver is synchronous, which means `can_send_packet_now` is NULL. However, in `gap_run_set_local_name` and `gap_run_set_eir_data`, `hci_reserve_packet_buffer` is used to reserve the HCI buffer. I cannot find a way to release the packet buffer under synchronous transport settings. Does that mean I must design the HCI driver as asynchronous?

mringwal, 2024-04-22:
Hi @AminoACID123 Please try the current develop branch. gap_run_set_local_name and gap_run_set_eir_data should work with synchronous implementations now as well.
In general, please consider if it's possible to implement an asynchronous driver.

AminoACID123, 2024-04-23:
I see. Thank you.

closed 2024-04-23 by AminoACID123