bluekitchen/btstack #550: Stopping and restarting Bluetooth would be useful

kevinjwalters, 2023-11-25:
**Is your feature request related to a problem? Please describe.**
I would like to be able to shutdown Bluetooth and then start it up later. 

My reason for doing this is due to the [Pi Pico W's odd incompatibility between DMA ADC and using Bluetooth](https://forums.raspberrypi.com/viewtopic.php?t=355701).

It looks like others may wish to shutdown Bluetooth for low power reasons https://github.com/raspberrypi/pico-sdk/issues/1479

**Describe the solution you'd like**
There's already a `btstack_run_loop_trigger_exit()` but I don't think this was intended to allow the Bluetooth stack to be shutdown/suspended _and then restarted_.

I'd like a some example code which uses Bluetooth then stops using Bluetooth triggered by the closing of a connection from another device and then after doing something unrelated starts up the Bluetooth stack again. I'd like that to work on the Pi Pico W. Happy to test there.

**Describe alternatives you've considered**
None.

**Additional context**
None.

peterharperuk, 2023-11-27:
I'll take a look at doing this in a pico-example

mringwal, 2023-11-27:
@peterharperuk great, thanks! Maybe you can also have a look at the forum post - my guess would be that there's no technical reason not to use CYW43/BTstack at the same time as DMA with ADC.

mringwal, 2024-01-10:
@peterharperuk Does it work? Happy New Year!

mringwal, 2024-06-19:
Closing due to lack of activity

closed 2024-06-19 by mringwal