bluekitchen/btstack #556: Compiler error: ESP-IDF 5.2

ricardoquesada, 2024-01-16:
```
/Users/ricardoq/progs/bluepad32/src/components/btstack/es8388.c:315:5: error: conflicting types for 'es8388_set_dac_ouput' due to enum/integer mismatch; have 'int(int)' [-Werror=enum-int-mismatch]
  315 | int es8388_set_dac_ouput(int output)
      |     ^~~~~~~~~~~~~~~~~~~~
In file included from /Users/ricardoq/progs/bluepad32/src/components/btstack/es8388.c:4:
/Users/ricardoq/progs/bluepad32/src/components/btstack/es8388.h:262:5: note: previous declaration of 'es8388_set_dac_ouput' with type 'int(es_codec_dac_output_t)'
  262 | int es8388_set_dac_ouput(es_codec_dac_output_t output);
      |     ^~~~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
[1090/1204] Building C object esp-idf/btstack/CMakeFiles/__idf_btstack.dir/src/hci.c.obj
ninja: build stopped: subcommand failed.
```

Using latest `develop` btstack branch.
Using ESP-IDF v5.2 branch.

Seems to be related to a new set of warnings/errors from GCC 13.

mringwal, 2024-01-17:
Thanks for reporting. (Hopefully) fixed on develop. Please reopen if build still fails.

ricardoquesada, 2024-01-17:
fixed. thanks for the quick fix

closed 2024-01-17 by ricardoquesada