bluekitchen/btstack #568: Mapping HID from tinyusb report to HID Classic Keyboard

emanuellopes, 2024-03-06:
**Is your feature request related to a problem? Please describe.**
Hi, I'm using tiny usb library with a keyboard and I'm sending the same report into Btstack.

For example the tinyusb report is like this {reportID, modifier, keycode1 keycode2, keycode3, keycode4, keycode5, keycode6}

But When I send this information to BTstack classic I need to send in this way
{0xa1, reportID, modifier, oem, keycode1 keycode2, keycode3, keycode4, keycode5, keycode6}

Do you have a structure to convert this or should I do manually?

Why do I need send 0xa1? do you have a link for documentation?

In case of consumer data, is only 3 values for example play/pause key {reportID, CD, 07}
I have this working on my side with the same approach 
{0xa1, reportID, CD, oem, 07, 0, 0, 0, 0, 0}

In case of consumer descriptor should I send the report in this way?

Thank you

mringwal, 2024-03-06:
Hi Emanuel. In Bluetooth HID, HID reports are sent as HID Protocol Messages which consists of a HIDP Message type and the payload. A1 is the message type for Input Reports. The BTstack function is called hid_device_send_interrupt_message(). It might be a good idea to add a little wrapper that's called 'hid_device_send_input_report()', which would then prepend the A1 to the payload. For now, please do this in your code.

Documentation: Chapter 3 in the Bluetooth HID Specification: https://www.bluetooth.com/specifications/specs/human-interface-device-profile-1-1-1/

closed 2024-03-06 by mringwal

emanuellopes, 2024-03-06:
Thank you for quicker response, helped me a lot