bluekitchen/btstack #572: hci_connection_timeout_handler never triggered

amorniroli, 2024-03-08:
**Describe the bug**

Not sure if it's a bug.

I've noticed that when HCI connection is created (function hci_connection_init), timer handler hci_connection_timeout_handler is set, but it's never triggered since timer it's not added to btstack timer (using btstack_run_loop_add_timer).

[image]

Is this correct?

mringwal, 2024-03-10:
Hi Alessandro. The hci_connection_init just initialized the hci connection timer. It is set for the first time on HCI_EVENT_CONNECTION_COMPLETE_EVENT here: https://github.com/bluekitchen/btstack/blob/272986f17af35a67815ab20897f6c91e710322a4/src/hci.c#L3733

Bluetooth itself does not define any kind of timeout on the HCI level (there are timeouts on higher level, e.g. during L2CAP setup). BTstack disconnects an idle HCI Connection == a connection without any L2CAP Channels after 10 seconds.

closed 2024-03-10 by mringwal

mringwal, 2024-03-12:
> Am I doing something wrong?

Yes :) 

The hci connection timeout is only used to discard the connection if it's not used. It's not related to the connection process.

To get shorter timeout for the connection attempt - called paging - , please call gap_set_page_timeout( timeout_in_625ns). The default is 0x6000 => 15 seconds

amorniroli, 2024-03-12:
> Yes :)
>
> The hci connection timeout is only used to discard the connection if it's not used. It's not related to the connection process.
>
> To get shorter timeout for the connection attempt - called [...]

Nice! I've missed that function. Thank you very much, now it's working as I was expecting : ) thank you very much