bluekitchen/btstack #577: too few arguments to function 'hci_event_create_from_template_and_arglist'

zenkoo, 2024-03-21:
../../chipset/cinnamon/controller.c:168:23: error: too few arguments to function 'hci_event_create_from_template_and_arglist'
  168 |     uint16_t length = hci_event_create_from_template_and_arglist(hci_outgoing_event, event, argptr);
      |                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from ../../chipset/cinnamon/controller.c:54:
../../src/hci_event.h:78:10: note: declared here
   78 | uint16_t hci_event_create_from_template_and_arglist(uint8_t *hci_buffer, uint16_t buffer_size, const hci_event_t *event, va_list argptr);
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

mringwal, 2024-04-02:
Thanks for reporting. It needed the buffer size. Fixed in 478f65a on branch develop.
Does it build for you now?

closed 2024-04-02 by mringwal

Nuts-Underline, 2024-11-27:
nop. having same error with  `hci_event_create_from_template_and_arglist` line 168

mringwal, 2024-11-27:
Sorry, we've missed that initially. It has been fixed on develop branch, in commit cf1c3df9.
Next release probably before Christmas.

Nuts-Underline, 2024-11-28:
switched to develop branch having other issue now #644