bluekitchen/btstack #581: Missing finalization of avdtp connection

amorniroli, 2024-03-29:
**Describe the bug**

AVDTP connection is never released if it'closed before stream has been established successfully..
**To Reproduce**

Steps to reproduce the behavior:

Just try this code snippet

```
uint8_t cid;
bd_addr_t address; // some address

a2dp_sink_establish_stream (address, &cid);
a2dp_sink_disconnect (cid)
a2dp_sink_establish_stream (address, &cid);  // returns ERROR_CODE_COMMAND_DISALLOWED
```

**Expected behavior**

The second call to a2dp_sink_establish_stream  should return ERROR_CODE_SUCCESS

**HCI Packet Logs**
Not needed

**Environment:  (please complete the following information):**
 - btstack branch develop (de13ba803acdb54c41ebfbca4e795757f719a2dd )

**Additional context**
My suggestion for a fix

[image]

mringwal, 2024-04-02:
Hi Alessandro. How did you run into this?

The main question here should be if it's allowed to call avdtp_disconnect before the event for avdtp_connect() was received in the first place. If yes, we need to make avdtp_connect() more robust, if not an avdtp_connect_abort() would be needed. 

Your suggestion might actually work, although I'd rather implement the 'abort' behaviour manually to make sure I didn't miss a case. Let me think about that a bit more.

mringwal, 2024-04-02:
Ok, please give 041b66c a try.

closed 2024-04-02 by mringwal

amorniroli, 2024-04-02:
Hi @mringwal ,

in my scenario,  the issue randomly occurred when calling hci_power_control (HCI_POWER_OFF) and a2dp_sink_establish_stream  had been already call but still not established.

For example:

- a2dp_sink_establish_stream
- hci_power_control (HCI_POWER_OFF) (sink still not established)
- hci_power_control (HCI_POWER_ON)
- a2dp_sink_establish_stream -> would fail for command disallowed

I'll try the fix in the afternoon, thank you ver much.

mringwal, 2024-04-02:
Did you then try to force disconnect by calling a2dp_sink_disconnect, and then run into this?

So, the next question is if/how to trigger avdtp finalize on power off. I would make the argument, that the establish stream request could stay pending and should get executed after the next power on.

amorniroli, 2024-04-02:
> Did you then try to force disconnect by calling a2dp_sink_disconnect, and then run into this?

Yes, extacly.

in my A2DP_HCI_PACKET_HANDLER, for HCI_STATE_HALTING btstack event I was trying to disconnect the A2DP connection, and then ran into the issue.

I confirm that b8bf1755110d8895952df06fbc69f4ab7788e64d  seems to resolve.