bluekitchen/btstack #585: binding for java is missing some classes

lgnativs, 2024-04-13:
BTStack class and some event not found when using binding for java

```
import com.bluekitchen.btstack.BTstack;
import com.bluekitchen.btstack.event.BTstackEventState;
import com.bluekitchen.btstack.event.GAPEventAdvertisingReport;
import com.bluekitchen.btstack.event.GATTEventCharacteristicQueryResult;
import com.bluekitchen.btstack.event.GATTEventCharacteristicValueQueryResult;
import com.bluekitchen.btstack.event.GATTEventNotification;
import com.bluekitchen.btstack.event.GATTEventQueryComplete;
import com.bluekitchen.btstack.event.GATTEventServiceQueryResult;
import com.bluekitchen.btstack.event.HCIEventDisconnectionComplete;
import com.bluekitchen.btstack.event.HCIEventLEConnectionComplete;
```

mringwal, 2024-04-13:
These classes are generated in platform/daemon/binding/java/gen when building everything with ant.
build.xml contains the target 'generate' which calls btstack/tool/java_binding.py

I've started a readme:
https://github.com/bluekitchen/btstack/blob/develop/platform/daemon/binding/java/README.md

closed 2024-04-13 by mringwal

lgnativs, 2024-04-13:
Thanks for your prompt reply and great work.

I will try to make this right away,thanks.