bluekitchen/btstack #587: L2CAP Connection Oriented Channels examples

Loc15, 2024-04-16:
Hi, I am writing a BTstack project with Pico W (RP2040), exactly I'm trying to do [this](https://github.com/rnconrad/WiimoteEmulator), but I get L2CAP_CONNECTION_RESPONSE_RESULT_REFUSED_SECURITY after HCI_EVENT_PIN_CODE_REQUEST where I reversed Wii's Bluetooth address. I don't know really if doing well the implementation of L2CAP channels or if even it's possible to do this project with btstack .

**To Reproduce**

```
int btstack_main(int argc, const char * argv[]){
    (void)argc;
    (void)argv;

    // allow to get found by inquiry
    gap_discoverable_control(1);
    // use Limited Discoverable Mode; Peripheral;
    gap_set_class_of_device(0x2504);
    // set local name to be identified - zeroes will be replaced by actual BD ADDR
    gap_set_local_name(hid_device_name);
    // allow for role switch in general and sniff mode
    gap_set_default_link_policy_settings( LM_LINK_POLICY_ENABLE_SNIFF_MODE );
    // allow for role switch on outgoing connections - this allow HID Host to become master when we re-connect to it
    gap_set_allow_role_switch(true);

    gap_set_required_encryption_key_size(7);

    // L2CAP
    l2cap_init();

    l2cap_register_service(l2cap_handler, PSM_SDP, 0xffff, gap_get_security_level());
    l2cap_register_service(l2cap_handler, PSM_HID_INTERRUPT, 0xffff, gap_get_security_level());
    l2cap_register_service(l2cap_handler, PSM_HID_CONTROL, 0xffff, gap_get_security_level());

#ifdef ENABLE_BLE
    // Initialize LE Security Manager. Needed for cross-transport key derivation
    sm_init();
#endif

    // SDP Server
    sdp_init();
    memset(hid_service_buffer, 0, sizeof(hid_service_buffer));

    hid_sdp_record_t hid_sdp_record = {
    	// hid sevice subclass 2504 Gamepad?, hid counntry code 33 US
    	0x2504,
        33,
        1,
        1,
        1,
        0,
        0,
        0xFFFF,
        0xFFFF,
        3200,
        wiimote_bt_report_descriptor,
        sizeof(wiimote_bt_report_descriptor),
        hid_device_name
    };

    hid_create_sdp_record(hid_service_buffer, sdp_create_service_record_handle(), &hid_sdp_record);
    btstack_assert(de_get_len( hid_service_buffer) <= sizeof(hid_service_buffer));
    sdp_register_service(hid_service_buffer);

    // See https://www.bluetooth.com/specifications/assigned-numbers/company-identifiers if you don't have a USB Vendor ID and need a Bluetooth Vendor ID
    // device info: BlueKitchen GmbH, product 1, version 1
    device_id_create_sdp_record(pnp_service_buffer, sdp_create_service_record_handle(), DEVICE_ID_VENDOR_ID_SOURCE_BLUETOOTH, BLUETOOTH_COMPANY_ID_BLUEKITCHEN_GMBH, 1, 1);
    btstack_assert(de_get_len(pnp_service_buffer) <= sizeof(pnp_service_buffer));
    sdp_register_service(pnp_service_buffer);

  	// HID Device
  	hid_device_init(1, sizeof(wiimote_bt_report_descriptor),wiimote_bt_report_descriptor);

       
    // register for HCI events
[... 258 lines cut ...]
[00:00:03.587] LOG -- l2cap.c.2374: connection complete con_handle 000b - for channel 20005D30 cid 0x0046
[00:00:03.597] EVT <= 0F 04 00 01 1B 04 
[00:00:03.600] EVT <= CD 09 0B 00 00 26 63 79 F7 6B D8 
[00:00:03.605] LOG -- sm.c.2263: LE Device Lookup: not found
[00:00:03.610] EVT <= CE 09 0B 00 00 26 63 79 F7 6B D8 
[00:00:03.616] EVT <= 0B 0B 00 0B 00 FF FF 8D FE 9B F9 00 80 
[00:00:03.620] CMD => 1C 04 03 0B 00 01 
[00:00:03.624] EVT <= 6E 00 
[00:00:03.627] EVT <= 13 05 01 0B 00 01 00 
[00:00:03.631] EVT <= 05 04 00 0B 00 13 
[00:00:03.634] LOG -- l2cap.c.1135: L2CAP_EVENT_CHANNEL_OPENED status 0x6a addr D8:6B:F7:79:63:26 handle 0xb psm 0x13 local_cid 00
[00:00:03.651] LOG -- l2cap.c.1152: ERTM mode 0, fcs enabled 0
[00:00:03.657] EVT <= 70 18 6A 26 63 79 F7 6B D8 0B 00 13 00 46 00 00 00 9B 06 A0 02 00 00 00 00 00 
channel open
L2CAP connection failed (cid 70, status 0x6a)
[00:00:03.671] LOG -- l2cap.c.2472: free channel 20005D30, local_cid 0x0046
[00:00:03.678] LOG -- l2cap.c.1410: l2cap_stop_rtx for local cid 0x46
[00:00:03.684] LOG -- l2cap.c.358: Stop Retransmission timer
[00:00:03.689] LOG -- l2cap.c.344: Stop Monitor timer
[00:00:03.694] LOG -- l2cap.c.1135: L2CAP_EVENT_CHANNEL_OPENED status 0x6a addr D8:6B:F7:79:63:26 handle 0xb psm 0x11 local_cid 00
[00:00:03.711] LOG -- l2cap.c.1152: ERTM mode 0, fcs enabled 0
[00:00:03.717] EVT <= 70 18 6A 26 63 79 F7 6B D8 0B 00 11 00 45 00 00 00 9B 06 A0 02 00 00 00 00 00 
```
**Environment: (please complete the following information):**
- Tested on RP2040 Pico W with 1.5.1 sdk

mringwal, 2024-04-17:
If you open the hci log in wireshark after converting it with tool/create_packet_log.py (see attached file), we can see 2 connection attempts. Weirdly, there's a role change event before the connection has been established. 200ms there's a connection complete, remote terminated.

The second is a bit better. The role change also comes before the successful connection complete - so this might be ok. Then, BTstack queries the remote features and also tries to open an SDP connection, but gets no result for the Read Remote Extended Features requests or gets a response for the L2CAP request, before the remote disconnect.

There's no PIN Request in the HCI log.

While BTstack can handle the 3 parallel L2CAP connections, I'd set them up one by one. Also, as this will be an HID Device, I would suggest to try the hid_keyboard / hid_mouse example although the behaviour should be the same. Can you try with a support USB Bluetooth dongle on a mac/linux/windows and port/libusb or port/windows-winusb?

There's no issue in BTstack. Feel free to post on the mailing list about your progress with the Wii.

[loc15-wii.pklg.zip](https://github.com/bluekitchen/btstack/files/15014675/loc15-wii.pklg.zip)

Loc15, 2024-04-18:
I don't gonna lie man, I'm little lost. I compile the [emulator](https://github.com/rnconrad/WiimoteEmulator) try to connect to wii but before of that, I erase all wii remote pairing pressing sync button on the console. I found that try to connect directly with the mac address the emulator doesn't work, it's doesn't connect. But if I put the emulator on listen for incoming connections mode it connect after you press sync button on the console, and after that I can to connect directly with the mac address. I got logs from wireshark for both cases.  
What would be a better way to emulate what is in the **search_wii** file with btstack?
Thanks for your help.
[connect_direct.pcapng.zip](https://github.com/bluekitchen/btstack/files/15031313/connect_direct.pcapng.zip)
[search_wii.pcapng.zip](https://github.com/bluekitchen/btstack/files/15031314/search_wii.pcapng.zip)

mringwal, 2024-04-19:
According to this Wii Emulator, the Wii must initiate the first connecting and start the pairing. So you need to make the Wii connect to your BTstack app. Before it connects, it will do an Inquiry and all it receives are the Class of Device and the Remote Name. In the log, they use CoD 0x2504 and change the name to "Nintendo RVL-CNT-01". You can set these two e.g. in the hid_keyboard_demo and try to connect from Wii. After that, you're almost done. 
I'm closing this as this it not an issue of the stack, just your application.

Again,  Feel free to post on the mailing list about your progress with the Wii. @ricardoquesada from bluepad32 might have some ideas as well.

closed 2024-04-19 by mringwal

ricardoquesada, 2024-04-19:
@Loc15 might help you:

From: https://wiibrew.org/wiki/Wiimote#Bluetooth_Pairing:

> If connecting by holding down the 1+2 buttons, the PIN is the
> bluetooth address of the wiimote backwards, if connecting by
> pressing the "sync" button on the back of the wiimote, then the
> PIN is the bluetooth address of the host backwards.

If you need to debug it from the "wii" side, as @mringwal mentioned, you can try running Bluepad32 on another device (picow,esp32,linux or macos) and see why it be sending incorrect packets.

Loc15, 2024-04-19:
@ricardoquesada I try on bluepad32 and it connect:

```
properties:
bp.bt.allow_en = false
No TLV for bp.bt.allowlist, returning default value
bp.bt.allowlist = <empty>
bp.ble.enabled = true
bp.gap.inq_len = 3
bp.gap.level = 2
bp.gap.max_len = 5
bp.gap.min_len = 4
bp.mouse.scale = 1.000000
No TLV for bp.version, returning default value
bp.version = '4.0.3'
bp.virt_dev_en = false
BR/EDR scan -> 1
BLE scan -> 1
posix_on_oob_event: Bluetooth enabled: 1
Device found: 28:CD:C1:06:5C:6B with COD: 0x002504, pageScan 1, clock offset 0x5a0d, rssi 234 dBm, name 'Nintendo RVL-CNT-01'
Creating device: 28:CD:C1:06:5C:6B (idx=0)
uni_bt_process_fsm, bd addr:28:CD:C1:06:5C:6B,  state: 4, incoming:0
uni_bt_process_fsm: Starting L2CAP connection
--> HCI_EVENT_CONNECTION_COMPLETE
SM_EVENT_IDENTITY_RESOLVING_STARTED
Identity resolving failed for 28:CD:C1:06:5C:6B

--> HCI_EVENT_LINK_KEY_REQUEST:
SSP User Confirmation Request with numeric value '926270'
SSP User Confirmation Auto accept
--> HCI_EVENT_AUTHENTICATION_COMPLETE_EVENT: status=0, handle=0x0001
L2CAP_EVENT_CHANNEL_OPENED (channel=0x0041)
PSM: 0x0011, local CID=0x0041, remote CID=0x0041, handle=0x0001, incoming=0, local MTU=1691, remote MTU=100, addr=28:CD:C1:06:5C:6B
HID Control opened, cid 0x41
uni_bt_process_fsm, bd addr:28:CD:C1:06:5C:6B,  state: 10, incoming:0
uni_bt_process_fsm: Create L2CAP interrupt connection
L2CAP_EVENT_CHANNEL_OPENED (channel=0x0042)
PSM: 0x0013, local CID=0x0042, remote CID=0x0042, handle=0x0001, incoming=0, local MTU=1691, remote MTU=100, addr=28:CD:C1:06:5C:6B
HID Interrupt opened, cid 0x42
Device 28:CD:C1:06:5C:6B is connected
posix: device connected: 0x55a29bbffe80
uni_bt_process_fsm, bd addr:28:CD:C1:06:5C:6B,  state: 12, incoming:0
uni_bt_process_fsm: starting SDP query
-----------> sdp_query_start()
Starting SDP VID/PID query for 28:CD:C1:06:5C:6B
Vendor ID: 0x057e - Product ID: 0x0306
Duplicate controller entry found for VID 0x0f0d PID 0x00ed (83,437)
Duplicate controller entry found for VID 0x146b PID 0x0d10 (100,101)
Duplicate controller entry found for VID 0x0e6f PID 0x02a1 (311,483)
Duplicate controller entry found for VID 0x0e6f PID 0x02a2 (312,484)
Duplicate controller entry found for VID 0x0e6f PID 0x02a5 (315,485)
Duplicate controller entry found for VID 0x0e6f PID 0x02a7 (317,440)
Duplicate controller entry found for VID 0x0e6f PID 0x02a8 (318,443)
Duplicate controller entry found for VID 0x0e6f PID 0x02bd (331,487)
Duplicate controller entry found for VID 0x0e6f PID 0x02bf (333,488)
Duplicate controller entry found for VID 0x0e6f PID 0x02c0 (334,489)
Duplicate controller entry found for VID 0x0e6f PID 0x02c6 (340,490)
Device detected as Wii controller: 0x23
uni_bt_process_fsm, bd addr:28:CD:C1:06:5C:6B,  state: 6, incoming:0
uni_bt_process_fsm: querying HID descriptor
Device 28:CD:C1:06:5C:6B does not need a HID descriptor, skipping query.
<----------- sdp_query_end()
uni_bt_process_fsm, bd addr:28:CD:C1:06:5C:6B,  state: 8, incoming:0
uni_bt_process_fsm: Device is ready
fsm: req_status
Device cannot connect in time, deleting:
	btaddr: 28:CD:C1:06:5C:6B
	bt: handle=1 (ACL), hids_cid=0, ctrl_cid=0x0041, intr_cid=0x0042, cod=0x00002504, flags=0x00003b00, incoming=0
	model: vid=0x057e, pid=0x0306, model='Wii', name='Nintendo RVL-CNT-01'
	battery: 0 / 255, type=unknown
	Wii: device 'Unknown', extension 'N/A'
Disconnecting device: 28:CD:C1:06:5C:6B
posix: device disconnected: 0x55a29bbffe80
Deleting device: 28:CD:C1:06:5C:6B
L2CAP_EVENT_CHANNEL_CLOSED: 0x0042 (channel=0x0042)
Couldn't not find hid_device for cid = 0x0042
L2CAP_EVENT_CHANNEL_CLOSED: 0x0041 (channel=0x0041)
Couldn't not find hid_device for cid = 0x0041
--> HCI_EVENT_DISCONNECTION_COMPLETE
^CCTRL-C - SIGINT received, shutting down..
```

but on the console... doesn't work

ricardoquesada, 2024-04-19:
mmm... I wouldn't k now then :(

ricardoquesada, 2024-04-19:
mmm... wait... it says "Device cannot connect in time, deleting".

That means that it couldn't finish the setup. Probably Wii console is sending you a request, and you are not answering it.

Loc15, 2024-04-19:
but it already connected right? because on wii console it even fail to create l2cap connection

ricardoquesada, 2024-04-19:
It connected at L2CAP level... but it didn't finish the setup.

But if you are having issues at L2CAP level, then I wouldn't know.

Loc15, 2024-04-19:
When bluepad send **fsm: req_status** and calls **uni_hid_device_send_intr_report** function should I get this data on L2CAP_DATA_PACKET event right ?

ricardoquesada, 2024-04-20:
yes, I guess... in the end it calls `l2cap_send()`, so it should trigger a `L2CAP_DATA_PACKET` ... but not 100% sure

Loc15, 2024-04-21:
> so it should trigger a L2CAP_DATA_PACKET

Yeah, that was what I thought, but no data arrives at that event. So weird.

Loc15, 2024-04-22:
@mringwal Hey man, one question, how to set a inquire access code ? I found this on internet : `hci_send_cmd(&hci_inquiry, HCI_INQUIRY_LIMITED, INQUIRY_INTERVAL, 0);` but after this the search stop.
I'm using the gap_inquiry.c example. I enabled DUT mode too before. I want to set IAC to 0x9E8B00.
Thanks.

mringwal, 2024-04-22:
you can call 

/**
 * @brief Set LAP for GAP Classic Inquiry
 * @param lap GAP_IAC_GENERAL_INQUIRY (default), GAP_IAC_LIMITED_INQUIRY
 */
void gap_inquiry_set_lap(uint32_t lap);

Loc15, 2024-04-24:
@mringwal Thanks mate.
I've been able to progress in the development, but now I'm getting this error **Ignore invalid report data packet, invalid size** and I can't receive data from the Wii. Do you know what I'm doing wrong ?
[log_invalid report.zip](https://github.com/bluekitchen/btstack/files/15087560/log_invalid.report.zip)

Loc15, 2024-04-25:
Fixed, it was a error in the HID descriptor

mringwal, 2024-04-26:
Excellent. My guess would have been: either your HID report had the wrong length, or your HID descriptor isn't valid/does not match.

Loc15, 2024-05-02:
@mringwal Everything was going smoothly, and then, a new error occurred. When opening a game or application, the Wii disconnect from the Wiimote, which it's normal. But when attempting to reconnect (following the same process to connect to the Wii before), the connection failed. I can't get complete wireshark file because it's too many info from the first connect until open a game then I just gonna show the part where the connection fails. I tried to connect doing `hid_device_connect`
[new_error.zip](https://github.com/bluekitchen/btstack/files/15182400/new_error.zip)

mringwal, 2024-05-06:
@loc15 In the log, you (re)connect to the Wii, start authentication. During this the link key stored in the first connection is used, but this does not seem to be the correct one. The difference is that BTstack has stored the link key. You could delete the link key before (re)connect using gap_delete_bonding(bd_addr) or even gap_delete_all_link_keys. Please check the Wii WiiBrew Wiki to see how re-connect works.

Loc15, 2024-05-14:
@mringwal Erasing pico's flash make it work and it reconnects but still doesn't work on homebrew, I guess it better than nothing. Thanks for all mate.