bluekitchen/btstack #594: Unexpected BNEP_EVENT_CHANNEL_CLOSED after ~40 sec.

western-hoolock, 2024-05-15:
I'm using btstack 1.6 on an ESP32. It's used to tether to an iPhone (iOS 17.5), the ESP32 takes on the PANU role.

Everything seems to work fine, I periodically (every 12 sec.) do an HTTP request from the ESP32 and get a response, but it seems like the iPhone/iOS insists on closing the connection after 40 sec. regardless of activity.
Without fail I get a ```BNEP_EVENT_CHANNEL_CLOSED``` in the BNEP handler around 40 sec. after the ```BNEP_EVENT_CHANNEL_OPENED``` event.

```
[00:00:06.555] LOG -- bnep.c.1169: BNEP_CONTROL: Type: 2, size: 3, is_extension: 0
[00:00:06.555] LOG -- bnep.c.936: BNEP_CONNECTION_RESPONSE: Channel established to FF:FF:FF:FF:FF:FF
[00:00:06.565] LOG -- bnep.c.116: BNEP_EVENT_CHANNEL_OPENED status 0x00 bd_addr: FF:FF:FF:FF:FF:FF, handler 0x400d3d2c
... This is where multiple successful HTTP requests are done.
[00:00:45.665] LOG -- l2cap.c.1200: L2CAP_EVENT_CHANNEL_CLOSED local_cid 0x42
[00:00:45.665] LOG -- bnep.c.1350: L2CAP_EVENT_CHANNEL_CLOSED cid 0x42, channel 0x3fff123c
[00:00:45.675] LOG -- bnep.c.1356: L2CAP_EVENT_CHANNEL_CLOSED state 4
[00:00:45.675] LOG -- bnep.c.152: BNEP_EVENT_CHANNEL_CLOSED bd_addr: FF:FF:FF:FF:FF:FF, handler 0x400d3d2c
```

Any ideas?

mringwal, 2024-05-15:
No obvious idea. Could you post the full HCI Packet Log after converting it with tool/create_packet_log.py?
Maybe something else is going on that causes the disconnect.

western-hoolock, 2024-05-15:
Here's the Wireshark packet log.
There's 4 successful HTTP requests in there, shortly after the last one succeeds, the channel is closed for some reason.
[channel_closed.zip](https://github.com/bluekitchen/btstack/files/15325842/channel_closed.zip)

mringwal, 2024-05-15:
I see. Nothing else is happening really. The disconnect is caused by a Link Layer time out, it would be different if iOS disconnects actively.

Only idea: disable BR/EDR Secure Connections by calling gap_secure_connections_enable(false) before connecting.

western-hoolock, 2024-05-15:
Looks like calling `gap_secure_connections_enable(false)` actually did the trick! It's been running for 10+ minutes now and it hasn't disconnected.

I'm not sure I understand why this works, was iOS waiting for a special response related to this security setting and because it never arrived the connection was closed?

Also, I don't know nearly enough about Bluetooth to understand what the (security) implications are of doing this. Can you elaborate a bit?

mringwal, 2024-05-18:
It's not related to iOS. In the past, there have been a number of issues in the BR/EDR Secure Connection implementation on the ESP32. Here's the latest one: https://github.com/espressif/esp-idf/issues/13661

The security implications are minor. As far as I know, the legacy pairing is not less secure than the newer one. It's just that the legacy one is a a custom mechanism that was created for Bluetooth only, while the new one uses standard cryptographic  algorithms (AES-CCM). Still, there hasn't been an academic paper that claimed that the older mechanism is insecure.

Could you retest your use case with the test libdtm_app from the other issue with BR/EDR Secure Connections enabled?
If it's fixed by that, please leave a comment there to let Espressif know. If not, please create a new issue and post logs files with the test library and comment that everything works without BR/EDR Secure Connections. Thanks!

western-hoolock, 2024-05-19:
Thanks for clarifying!

I tried to build with the test lib in espressif/esp-idf#13661 but the linker complains about 2 undefined references, likely because this project is still on esp-idf 4.x. I won't be upgrading this project to v5 anytime soon since it's quite involved because of the netif changes in esp-idf v5.

This project ran fine the last couple of years by the way, using BTStack 1.3.2, it was only because apparently after a certain iOS update it refused to connect that I started investigating. Tried upgrading to BTStack 1.6, but that didn't improve things until I tried your ```gap_secure_connections_enable(false)``` suggestion. Then I added that call to the version that still used BTStack 1.3.2 and it also started working again!

I'll revisit this if I ever come around to upgrading this project to esp-idf v5.

youngitil, 2024-06-08:
how to use ESP32 run as a NAP role?