bluekitchen/btstack #597: Examples with address parsing logic seems wrong

steven-buytaert, 2024-05-23:
It seems there are several examples, that need to parse an address that is passed on the command line, apply the wrong logic.

The code usually found is this:

`    int arg = 1;
    cmdline_addr_found = 0;
    
    while (arg < argc) {
        if(!strcmp(argv[arg], "-a") || !strcmp(argv[arg], "--address")){
            arg++;
            cmdline_addr_found = sscanf_bd_addr(argv[arg], cmdline_addr);
            arg++;
            if (!cmdline_addr_found) exit(1);
            continue;
        }
       // usage called or something 
        return 0;
    }`

But this logic is wrong since when the if clause is not taken, the argc is not advanced and the program loops forever.

The proper template should look like this if I am not mistaken:

`  int arg;

   for (arg = 1; arg < argc; arg++) {
         if(!strcmp(argv[arg], "-a") || !strcmp(argv[arg], "--address")){
            arg++;
            cmdline_addr_found = sscanf_bd_addr(argv[arg], cmdline_addr);
            if (!cmdline_addr_found) {
                usage(argv[0]);
                exit(1);
            }
        } 
   }
`

This is at least the case for gatt_battery_query.c, le_streamer_client.c, le_credit_based_flow_control_mode_client.c, gatt_heart_rate_client.c, gatt_device_information_query.c and gatt_browser.c.

Do I need to make a pull request for this? It would be my first on Github.

Kind regards,

Steven

steven-buytaert, 2024-05-24:
Hi Matthias,

I've send you a mail with the patch and some more info.

Cheers,

Steven

mringwal, 2024-05-27:
Fixed on develop branch.

closed 2024-05-27 by mringwal