bluekitchen/btstack #599: Fails to compile on ESP32C6

ricardoquesada, 2024-06-07:
**Describe the bug**

BTstack 1.6.1 fails to compile on ESP32C6 using ESP-IDF v5.1.4

**To Reproduce**

```
# go to a project that uses BTstack
idf.py set-target esp32c6
idf.py build
```

**Expected behavior**

It should compile Ok

** Logs **

```
/app/ricardoquesada/esp-idf-arduino-bluepad32-template/components/btstack/btstack_audio_esp32_v5.c: In function 'btstack_audio_esp32_init':
/app/ricardoquesada/esp-idf-arduino-bluepad32-template/components/btstack/btstack_audio_esp32_v5.c:97:32: error: 'GPIO_NUM_35' undeclared (first use in this function); did you mean 'GPIO_NUM_30'?
```

**Environment:  (please complete the following information):**
 - Current BTstack branch: `master`, tag: `1.6.1`
 - Bluetooth Controller: ESP32-C6
 - Remote device: N/A

**Additional context**
https://github.com/ricardoquesada/esp-idf-arduino-bluepad32-template/actions/runs/9410909727/job/25923329434

mringwal, 2024-06-07:
Hi Ricardo. Do you know of any ESP32 C6 Devkits that have audio? If yes, we could pre-configure the audio driver for that setup. If not, we could/need to disable audio for the C^.

ricardoquesada, 2024-06-08:
I wouldn't know. Im not using the C6 for audio.
I got this one just to the test the BT gamepads part:
https://docs.espressif.com/projects/espressif-esp-dev-kits/en/latest/esp32c6/esp32-c6-devkitc-1/index.html

Not sure whether they have any audio features.

Would it be possible to change the API or add a new API in a way that the user has to specify the GPIOs to use ?
(to avoid hardcoding the GPIOs )

mringwal, 2024-06-12:
We'll try to change the logic to a whitelist with defaults and disable audio otherwise to avoid compile errors like this.

mringwal, 2024-06-14:
Hi Ricardo. Could you check the latest develop branch?  In https://github.com/bluekitchen/btstack/commit/366f7a2538e77b75de9c19704bd4204a3559452d we've made the audio part option, so you should be able to disable it (comment `CONFIG_BTSTACK_AUDIO`)

ricardoquesada, 2024-06-14:
Thanks for the quick fix. Currently on vacation. I'll test it next week.

ricardoquesada, 2024-06-18:
By disabling the I2S driver from menuconfig, it compiles ok. ty!

closed 2024-06-18 by ricardoquesada

mringwal, 2024-06-18:
Great. Now, there should be no problems with new ESP32 MCUs for projects that don't use audio.

ricardoquesada, 2024-06-18:
Filed this new bug: https://github.com/bluekitchen/btstack/issues/611