bluekitchen/btstack #600: BT Sound bar and PS4 controller bandwidth clash

XZCE, 2024-06-09:
Hi,

I'm using this brilliant library with my Pi Pico W (tried both the RasPi Pico SDK version and the latest version here) with my Pico overclocked to 250MHz.  I've got a project that works "just fine" with either a BT speaker, or, with a PS4 gamepad...  Everything is fine, except when I have both devices going at the same time - then the sound becomes choppy.

I've tried the same scenario in Windows 8.1 - the sound is never choppy, but I do see the PS4 gamepad sometimes glitches on that machine, it seems to hang on to the current button state longer than it should...  But this is how I'd like my Pi Pico W project to work - I.E. give priority to the BT speaker when the BT bandwidth is full with 2 noisy devices going at the same time.

Is there some setting I can change to make this happen?  And thanks for this library!  It's been pretty easy to work with!

mringwal, 2024-06-10:
While there are mechanisms to tweak bandwidth per connection, it's not easy to use.

In your case, the CYW4343W should be Central for both connections and be able to schedule it's A2DP Source role to the speaker as well as the HID Host to the PS4 Controller without problems. Both speaker as well as PS4 probably initiate the connection, but are willing to enter Peripheral Role.

Please make sure that this happens. See example/hid_host.c and example/a2dp_source.c as comparison
Basically, you'll need this:

    // Allow sniff mode requests by HID device and support role switch
    gap_set_default_link_policy_settings(LM_LINK_POLICY_ENABLE_SNIFF_MODE | LM_LINK_POLICY_ENABLE_ROLE_SWITCH);

    // try to become master on incoming connections
    hci_set_master_slave_policy(HCI_ROLE_MASTER);

Maybe only the second one is sufficient - check if the role change happens during incoming connection via the packet log.

You can add/remove the sniff mode for testing

XZCE, 2024-06-11:
Thanks!  I do have those lines in my code (it's basically the BT PicoW PS4 controller example someone else did on GitHub, combined with your a2dp_source_demo.c - but I've also got some of your hid_host_demo.c code in there too, for connecting an old BT3 keyboard - but I'm not connecting that in my testing here).  I've really only noticed issues with the GamePad and SoundBar together.  The keyboard is really quite quiet, only sending reports when a key is pressed or released.

Actually, I've also jammed in some of the "gatt" code in there too, for connecting a BT5 keyboard, which all works - except I can only send one set LED status message, where I've commented out the "if (client->state != HIDS_CLIENT_STATE_CONNECTED) return ERROR_CODE_COMMAND_DISALLOWED;" in hids_client_send_write_report() because I think it's stuck in HIDS_CLIENT_W4_REPORT_DONE.  Maybe I'm missing some code in my project that resets "done" back to "connected"?  hog_host_demo.c doesn't seem to do that in "handle_gatt_client_event" but I see the library has another static function of the same name that does it in hids_client.c.  Anyway, keyboard LEDs are all working for me now, and I don't think this is related to my actual bandwidth issue.

I tried removing the sniff policy too, but I then never get any reports from the GamePad.  So I think the code is about right in that area.

I also wrote some extra code to report how many GamePad reports I get every second, it's about 30 (unless I move the analog sticks and it goes into the hundreds)...  But I do notice when I connect the BT speaker, the GamePad reports go to about 60 a second, then if I disconnect the BT speaker, it returns to about 30 again.  This confuses me, unless the GamePad is picking up signals from the Pico meant for the BT speaker?

XZCE, 2024-06-12:
Ah, OK, I don't think I see that happening in the log.  (Attached.)
[hcilog.txt](https://github.com/user-attachments/files/15800272/hcilog.txt)

mringwal, 2024-06-12:
Well, it doesn't mention "role change", but we need to look at the actual HCI Packets. When you get that, you can run tool/create_packet_log.py and open the file with Wireshark.

XZCE, 2024-06-12:
Does this look right?  I'm using Earl's Arduino-Pico example for packet logging here...
[hcipackets.txt](https://github.com/user-attachments/files/15802319/hcipackets.txt)

XZCE, 2024-06-13:
I've found time to go through the converted .txt logs in Wireshark - they seem mostly reasonable, except sometimes I'm getting interleaved output.  Looking at Earl's log_packet code, [BT @xxxx] should always be at the start of a line, but when I connect the 2 devices it sometimes isn't?  Interrupt perhaps?  I can say this is all running on Core 0 - because Core 1 is dedicated to servicing a bus connection to a vintage computer to emulate RAM and other things.  Anyway, Wireshark is saying the Pico is master in all the messages I've looked at.

Here's another log, with the PS4 Gamepad connected first, then the BT speaker:
[hcipackets2.txt](https://github.com/user-attachments/files/15815222/hcipackets2.txt)

Unfortunately, it's the same result, choppy sound.  Actually, looking at the logs, where the corruption (interleave) occurs, is where I can see packets being sent to the BT speaker have missed a few hundred milliseconds - not sure if that helps though, probably just the log file corruption?

XZCE, 2024-06-13:
I've been trying a lot of things without too much success, but I've got my BT headphones connected at 16KHz (I was using 32KHz for the sound bar, but that cheap device doesn't produce any sound when connected at this lower frequency - the headphones were much more expensive, so maybe they implement BT sound in a more complete way)...

The headphones work much better at this lower speed, they had the exact same issue at 32KHz, but eventually also fail after a while, with the same choppy sound.  It took a few minutes, working pretty well before that happened.

One thing I'm doing though, is configuring the Pico to have a single PWM interrupt going at 16KHz (it was 32KHz before) which runs some code to fill the buffers - it also powers a mono-sound-circuit which is external to BT, charging a capacitor based on consumed data bytes.  Maybe, and I'm speculating here, the 32KHz interrupt time I'm using is too much for the BT library to be stable with multiple devices connected?

I'll try to come up with a minimal example, when I get time, but I can say for sure that the mono-sound output has no audible glitches like the BT sound devices - so I'm not at all confident this will help at all in my quest to get this all working nicely together.  I'm pretty sure the Raspberry Pi guys didn't expect me not to have a PWM interrupt configured when I'm using BT?  (It's a different device with its own CPU after all, isn't it?)

mringwal, 2024-06-14:
With logging enabled at 115200, A2DP will get too slow. The main idea here was to verify that the role change happens and that you end up being master to both devices. 

Not sure how these logs are generated and why they are diff from our hci_dump_embedded_stdout.c
Please switch to that or extend the tools/create_packet_log.py to handle the new format. with the [BT @10792] timestamp.

You can also reduce the audio qualify by specifying a lower 'bit pool' (which reduces the audio bandwidth).

If you're using other feature, I cannot really comment, but you could try to avoid these while looking for the root case, e.g. by playing a sine tone or using the hxcmod player.

XZCE, 2024-06-14:
I've replicated it "out of circuit" with the Pico just on its own (with 10 fixed pre-set tone buffers, which for each, are set to "available" at the just the right time) and the Pico (which is host) is basically doing nothing else on my end other than saying "ready".

1/ PWM at 32KHz will run the BT sound bar via the interrupt (counting to 128 stereo samples) indefinitely.
2/ Connect the PS4 Gamepad and it dies sooner or later, mostly sooner.
3/ Fixed 4ms test loop to set the next buffer to "available" also runs it indefinitely.
4/ Connect the PS4 Gamepad and it dies.
5/ Same as 3, but tries to keep track of losses, except if the tight-loop I'm using gets 7ms out of sync. and I reset.
6/ Sadly, also no good.

The 4ms time test was calculated from the audio devices being set to 32KHz and my 2 BT audio devices I have needing 128 stereo samples for each sub-buffer, the BT sound bar needing 5 for the total buffer size it supports.  My headphones accept 7 sub-buffers.  I assume this is all normal for BT audio.

I'll try some other changes, as you suggest, but at this point I'm feeling like I'd need a new example "combined PS4 controller & A2DP audio" because I'm not using any other features so I don't know what else I can do.  I've simply mashed together 2 working examples without success.  PS4 controllers are pretty popular, and it seems BT sound devices (since I've tried 2 with the same result) don't seem all that important with the issue here.

Thanks for your assistance on this, it's appreciated, but I don't think I'm going to get this working myself.

mringwal, 2024-06-14:
Totally unrelated. I don't understand the role of the PWM here. BTstack is in a2p_source role and send audio to a Bluetooth speaker while it receives HID reports from the PS4. 

The main question here is if the Bluetooth Controller (CYW43439) supports both connections without problems, and/or, if you have an issue with your application. You could try to combine the examples on a desktop (see port/libusb), where you can get HCI logs easily. In these logs, the timestamps would tell if an audio packet is sent roughly every 20 ms and if HID reports are received continously.

XZCE, 2024-06-14:
Ah, the PWM simply generated mono-sound for my project @32KHz (as described earlier) and I added those bytes (short ints, doubled up for stereo) to the BT library buffers - that's all.  It was also a nice way to get the sub-buffer timing right, rather than using the example which had to keep track of "acc_num_missed_samples".  But I've removed that completely from my tests now, and tried 2 other ways to keep these pre-generated audio packets going to the BT audio device, and sadly, both failed (but only when the PS4 controller is connected).

But this is no longer my application, it's all example code now.  I understand what the a2dp was doing to get the buffers ready to send at the right time, I'm trying my best to do the same, but it doesn't work for me on the Pico.

XZCE, 2024-06-15:
After a lot of trials, I can confidently say the PS4 controller will produce over 650 HID reports a second (just sitting there, with no buttons pressed or analog sticks wiggled, I don't know why it thinks it needs less than 2ms response time for these messages).  This is too much, and that's the problem.  With the Pico spending so much time in the BT library via its interrupt, it doesn't have much time left in my main loop (which serves the BT audio) and it's constantly blocking the "buffer fill" logic (which simply sets a flag, since the buffers are pre-filled with a constant tone).  My real project is actually more resilient, because of the PWM interrupt - but it's not immune, and eventually succumbs too.

The only solutions I can see is to somehow tell the PS4 controller not to send so many HID reports, or for this library not to bother processing all of them.  You could throw 4 out of 5 away and not notice.  Then again, as I wrote before, I've seen the HID reports come at around 30 a second - and you wouldn't want to throw any away.  But it really seems to be the BT speaker connected that's increasing the frequency of PS4 HID reports here.

I actually have the PS4 controller working via wired USB too - where each HID report that arrives triggers the next HID report request - so I get less than 500 a second - always.  This would be better than the "automatic" HID reports that BT seems to use.

Finally, I have to report, it's actually really hard to get both devices connected (in pair mode, at least) and I suspect with the PS4 controller making so much noise, it's not leaving enough radio bandwidth for a 2nd connection to be easy.  But I know next to nothing about BT, so I wouldn't really know what's going on - these 2 devices pair quite well individually.  But I will bow out here, I have no more ideas and I'll just accept I can use one or the other but not both.

XZCE, 2024-06-16:
On my last point (hard to connect 2 devices) it's actually only hard when the PS4 controller is paired first (every time I reflash the Pico, I obviously need to pair again)...  If I pair the BT speaker first, then the PS4 controller has no issue at all pairing/connecting.  If I set the BT MAC address to a keyboard (instead of the PS4 controller) and pair/connect that first, the BT speaker has no issue connecting.  It's only when the PS4 controller is paired first and sending HID reports that there's an issue.  I've looked in the HCI log output, but all I see are the two same packets over and over (ACL & CMD) which appear to be the HID report coming in (and I don't know, but maybe the CMD is an immediate request for another one - which maybe I would like to slow down!).  See attachment.

If I pair the BT speaker first, then connect the PS4 controller, I can power-down the BT speaker and reconnect (in pair mode or not) without any issues - as many times as I like.  It's very confusing.

Is there an example that shows how to connect multiple devices?  Previously I was just sending connect commands for all the BT MAC addresses I want to connect when BTSTACK_EVENT_STATE goes to HCI_STATE_WORKING.  I've now changed that to a round-robin algorithm, which attempts to connect one device only and waits for it to fail before trying the next.  This is how I discovered that the PS4 controller blocked new connections.

I still think this library is incredibly neat too, thanks again.

[hcilog.txt](https://github.com/user-attachments/files/15856195/hcilog.txt)

XZCE, 2024-06-16:
I nearly forgot to mention, that when I change "#define AUDIO_TIMEOUT_MS 10" to be 2 instead, things improve a lot, but still I get a gap in audio every couple of minutes (only when the PS4 controller is connected).

My version of a2dp_demo_audio_timeout_handler() is only picking up the flags set by my PWM interrupt though, so I can call it as often as I like.  It will only send the pre-generated tone buffers "at the right time" based on them being precisely set by that interrupt, every 128 stereo samples.

So, some progress, but still not fully working.

XZCE, 2024-06-16:
> You can also reduce the audio qualify by specifying a lower 'bit pool' (which reduces the audio bandwidth).

This advice has actually helped a lot, thank you.  I'm closing this issue, but it would be really good if you could find the time to borrow a PS4 controller and try it out yourself...  It would be great not to have to reduce quality.

closed 2024-06-16 by XZCE