bluekitchen/btstack #601: hid_host_demo:  discoverable and connectable by default

Slion, 2024-06-10:
I was trying to use hid_host_demo on Arduino Pi Pico W without `HAVE_BTSTACK_STDIN` from a device that must initiate connection. 
It took me some time to find out that I needed the following calls after the l2cap init to be able to see the Pico W host from my device.

```
    gap_connectable_control(1);
    gap_discoverable_control(1);
```

Maybe be could also have a configuration option in that demo like `AUTO_CONNECT_TO_HARD_CODED_DEVICE_WHEN_NO_STDIN` so that this could be disabled too.

mringwal, 2024-06-12:
Hi there. You don't need the gap_connectable_control, as that's automatically enabled if your application registers any (e.g. HID) services.

If `HAVE_BTSTACK_STDIN` is not defined, the hid_host_demo will initiate a connection to the device with remote_addr_string. 

Isn't that what you're asking for?

(With `HAVE_BTSTACK_STDIN`, you can trigger a device discovery and then connect to a found device).

Slion, 2024-06-13:
>  You don't need the gap_connectable_control, as that's automatically enabled if your application registers any (e.g. HID) services.

I was wondering about that too and did not get around testing it. Good to know, I'll try without it now.

> (With HAVE_BTSTACK_STDIN, you can trigger a device discovery and then connect to a found device).

How do I trigger it? Maybe I'm being blind here but I can't see the discovery in the code.

> If HAVE_BTSTACK_STDIN is not defined, the hid_host_demo will initiate a connection to the device with remote_addr_string.

In may case I want the device to initiate the connection with the Pico W and that's why I needed it to be discoverable. I guess what I'm saying is the demo should be discoverable by default like hosts usually are.

Slion, 2024-06-13:
I can confirm `gap_connectable_control` is not needed to initiate a connection from the device, only `gap_discoverable_control` was needed in my case for the device to find the host.

mringwal, 2024-06-13:
Hi

Sorry, the discovery wasn't in the hid_host_demo, there's one in the a2dp_source_demo :)
In hid_host_demo, you can press 'a' to connect to a pre-defined remote.

Your use case is a bit special. In general, keyboards don't have any kind of UI which would allow to select a HID Host.

I'd like to point out that while Bluetooth devices have traditionally been discoverable by default, in newer times, they are not discoverable by default for privacy reasons. At least iOS, Android, macOS are not discoverable without entering their settings or similar. I don't know what's the default on Windows but I would expect it to be similar.

Nevertheless, I don't mind hid_host_demo to be discoverable, so I've updated the demo on the develop branch.

closed 2024-06-13 by mringwal

Slion, 2024-06-13:
> In hid_host_demo, you can press 'a' to connect to a pre-defined remote.

That's weird I'm not seeing this in the code and it does not work for me. AFAICT tell the demo does not do a scan. Is there an example for Classic scan somewhere?

> Your use case is a bit special. In general, keyboards don't have any kind of UI which would allow to select a HID Host.

Are you telling me your keyboard does not have a screen? 😁 Yeah, it is a weird use case. I'm not using physical keyboards anymore. I use [giant Android tablets](https://slions.net/resources/interface.14/) instead. Been like that for about 18 months already. There is no going back for me I'm afraid.

Slion, 2024-06-13:
> In hid_host_demo, you can press 'a' to connect to a pre-defined remote.

Ok that's definitely not in the `hid_host_demo` so it does not look like that demo can connect to any device outside of the one that's hard-coded.

However it looks like this is indeed a feature of the `a2dp_source_demo` so I guess I should look at that code to discover HID devices.

mringwal, 2024-06-14:
> I use [giant Android tablets](https://slions.net/resources/interface.14/) instead.

Nice! Pierre and I provided a [macOS version](https://www.houdah.com/type2Phone/) until Apple limited the Bluetooth APIs on macOS to behave rather weird.

Slion, 2024-06-14:
> At least iOS, Android, macOS are not discoverable without entering their settings or similar.

Android devices are indeed hidden by default. An application can ask the user to make it visible but will reset back to hidden automatically after 2mn. Also I'm under the impression that when you enter Bluetooth settings it becomes visible too, could be wrong though. There is no visibility toggle option in the settings anymore.