bluekitchen/btstack #603: gap_inquiry_start from hid_host_demo never has rssi and name

Slion, 2024-06-13:
I ported the inquiry from `a2dp_source_demo` to `hid_host_demo` and it does find some devices but weirdly none have names or rssi like they do in `a2dp_source_demo`. Any tips?

mringwal, 2024-06-14:
There are three levels of inquiry response. The basic one does have neither name nor rssi. Please check packet log.
Maybe you need to enable higher levels with: `hci_set_inquiry_mode(INQUIRY_MODE_RSSI_AND_EIR)`
See gap_inquiry which also stores device addresses in a list and does a remote name query if needed.

Slion, 2024-06-14:
> Maybe you need to enable higher levels with: hci_set_inquiry_mode(INQUIRY_MODE_RSSI_AND_EIR)

That's it, that's what the a2dp source demo uses too. It did the trick, getting a lot more names now.
I'll look at gap_inquiry too, thanks for the tips.

closed 2024-06-14 by Slion

Slion, 2024-06-15:
> See gap_inquiry which also stores device addresses in a list and does a remote name query if needed.

I see now you meant to point me to that inquiry example rather than a function 🤦
Just tried it and that works great, could even fetch my Logitech K810 name using remote name query 🥳

mringwal, 2024-06-19:
Glad you got it working. What project are you working on?

Slion, 2024-06-19:
> Glad you got it working.

You got it working, I'm just trying to understand how it works 😁

> What project are you working on?

Mostly assessing the Bluetooth HID capabilities of the Pico W for various projects.