bluekitchen/btstack #604: hid_host_demo: Logitech Keyboard K810 HID descriptor not available

Slion, 2024-06-13:
Pico W SDK 1.5.1

Connecting to Logitech K810 keyboard results in `Cannot handle input report, HID Descriptor is not available, status 0x11` which is `ERROR_CODE_UNSUPPORTED_FEATURE_OR_PARAMETER_VALUE`. Though the connection is actually successful and if you mark the descriptor has being available it works properly somehow.

mringwal, 2024-06-14:
Please (always) attach HCI packet logs.

Slion, 2024-06-14:
> Please (always) attach HCI packet logs.

Sorry about that, I need to look into how to get those.

peterharperuk, 2024-06-14:
Does this help? https://github.com/raspberrypi/pico-examples/blob/develop/pico_w/bt/README.md

Slion, 2024-06-16:
Logs from pairing with Logitech K810 with a connection initiated from our Pico W host:
[logitech-k810-pairing.zip](https://github.com/user-attachments/files/15856417/logitech-k810-pairing.zip)

Slion, 2024-07-06:
I think I can't reproduce that now. It could be because I switched to the develop branch or it could just be because I increased the size of the attribute buffer through `MAX_ATTRIBUTE_VALUE_SIZE`.

closed 2024-07-06 by Slion