bluekitchen/btstack #611: Failed to compile when target is esp32-c6: undefined reference to 'esp_vhci_host_send_packet'

ricardoquesada, 2024-06-18:
Try to compile a BTstack based project for ESP32-C6.

Do:

- `idf.py set-target esp32-c6 && idf.py build`

Output:

```
/Users/ricardoq/.espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/12.2.0/../../../../riscv32-esp-elf/bin/ld: esp-idf/btstack/libbtstack.a(btstack_port_esp32.c.obj): in function `transport_send_packet':
/Users/ricardoq/progs/bluepad32/src/components/btstack/btstack_port_esp32.c:298: undefined reference to `esp_vhci_host_send_packet'
/Users/ricardoq/.espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/12.2.0/../../../../riscv32-esp-elf/bin/ld: esp-idf/btstack/libbtstack.a(btstack_port_esp32.c.obj): in function `transport_can_send_packet_now':
/Users/ricardoq/progs/bluepad32/src/components/btstack/btstack_port_esp32.c:288: undefined reference to `esp_vhci_host_check_send_available'
/Users/ricardoq/.espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/12.2.0/../../../../riscv32-esp-elf/bin/ld: esp-idf/btstack/libbtstack.a(btstack_port_esp32.c.obj): in function `transport_open':
/Users/ricardoq/progs/bluepad32/src/components/btstack/btstack_port_esp32.c:257: undefined reference to `esp_vhci_host_register_callback'
collect2: error: ld returned 1 exit status
```

This issue might be relevant: https://github.com/espressif/esp-idf/issues/12194

ricardoquesada, 2024-06-19:
Using ESP-IDF 5.1.4

mringwal, 2024-06-19:
Thanks for digging out the similar issue. It is indeed the same problem. I've raised it again, as it hasn't been fixed.
https://github.com/espressif/esp-idf/issues/14018

mringwal, 2024-06-20:
Espressif confirms that VHCI doesn't work on ESP32-C6 at the moment and that they plan to fix it.

If you're in a hurry, you can try to enable Bluedroid in sdkconfig (CONFIG_BT_BLUEDROID_ENABLED) and remove the #error guard in BTstack's main.c template (you'll get an error like 'Different Bluetooth Host stack selected -..' ). If you try, please let me know if this works, I don't have a C6 to test with (only classic ESP32, C3 and S3 at the moment).

ricardoquesada, 2024-06-20:
Thanks for looking at it.

Kind of busy these days on other projects, but I find some time I'll do the test and report my findings here.

Thanks!

ricardoquesada, 2024-06-21:
I was able to test it.

- Using ESP-IDF 5.1.4
- BTstack develop + Bluepad32
- `idf.py set-target esp32-c6`
- `menuconfig -> components -> Bluetooth -> Host (Bluedroid)`
- `idf.py build`

It builds Ok. No error from BTstack.

But Bluetooth is not working... or at least I'm not seeing any packet.

mringwal, 2024-06-21:
The error I was expecting is from our template main.c, which you might have replaced / changed.

Could you check if you have Bluetooth enabled (`CONFIG_BT_ENABLED`)?

I've ordered ESP32-C6 and ESP32-H2 (don't remember hearing about those before).

ricardoquesada, 2024-06-22:
Yes, I confirm it was enabled:

```
CONFIG_BT_ENABLED=y
CONFIG_BT_BLUEDROID_ENABLED=y
CONFIG_BT_CONTROLLER_ENABLED=y
```

ricardoquesada, 2024-09-07:
according to https://github.com/espressif/esp-idf/issues/14018 it already fixed.

But I tried with latest esp-idf v5.1 branch  + btstack develop  + bluepad32, and it compiles Ok, but it seems that it is not receiving any packet.

Not sure whether something needs to be done in btstack code... but I tried with:

- esp32-c6: compiles, but cannot see any BLE packet
- esp32-h2: compiles, but cannot see any BLE packet
- esp32-c3: works ok

mringwal, 2024-09-07:
Thanks for testing. I didn't get around to that yet. 

Would you have a packet log for the c6 and/or h2? 
(The dev kits are here now somewhere, with a P4 in the mail)

ricardoquesada, 2024-09-18:
(sorry for the delay... i've been busy..)

This is the log with HCI dump enabled.
As you can see, "nothing" is received.

What I tried was to connect using  BLE gamepad, but no packets are received

```
I (50) boot: ## Label            Usage          Type ST Offset   Length
I (58) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (65) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (73) boot:  2 factory          factory app      00 00 00010000 00100000
I (80) boot:  3 coredump         Unknown data     01 03 00110000 00010000
I (88) boot: End of partition table
I (92) esp_image: segment 0: paddr=00010020 vaddr=42080020 size=23680h (145024) map
I (129) esp_image: segment 1: paddr=000336a8 vaddr=40800000 size=0c970h ( 51568) load
I (141) esp_image: segment 2: paddr=00040020 vaddr=42000020 size=7e00ch (516108) map
I (244) esp_image: segment 3: paddr=000be034 vaddr=4080c970 size=05d2ch ( 23852) load
I (250) esp_image: segment 4: paddr=000c3d68 vaddr=408126a0 size=02438h (  9272) load
I (257) boot: Loaded app from partition at offset 0x10000
W (284) clk: esp_perip_clk_init() has not been implemented yet
I (291) cpu_start: Pro cpu start user code
I (291) cpu_start: cpu freq: 160000000 Hz
I (292) cpu_start: Application information:
I (294) cpu_start: Project name:     bluepad32_esp32_example_app
I (301) cpu_start: App version:      4.1.0-14-gf4c4d9f-dirty
I (307) cpu_start: Compile time:     Sep 17 2024 18:06:45
I (313) cpu_start: ELF file SHA256:  4d9b4e16a35ae1a3...
I (319) cpu_start: ESP-IDF:          v5.1.4-828-g33fbade6b8
I (325) cpu_start: Min chip rev:     v0.0
I (330) cpu_start: Max chip rev:     v0.99
I (335) cpu_start: Chip rev:         v0.0
I (340) heap_init: Initializing. RAM available for dynamic allocation:
I (347) heap_init: At 4081FC90 len 0005C980 (370 KiB): D/IRAM
I (353) heap_init: At 4087C610 len 00002F54 (11 KiB): STACK/DIRAM
I (360) heap_init: At 50000000 len 00003FE8 (15 KiB): RTCRAM
I (367) spi_flash: detected chip: generic
I (371) spi_flash: flash io: dio
W (375) spi_flash: Detected size(8192k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
I (388) sleep: Configure to isolate all GPIO pins in sleep state
I (395) sleep: Enable automatic switching of GPIO sleep configuration
I (402) esp_core_dump_flash: Init core dump to flash
I (408) esp_core_dump_flash: Found partition 'coredump' @ 110000 65536 bytes
I (415) coexist: coex firmware version: 27d8387
I (421) coexist: coexist rom version 5b8dcfa
I (426) app_start: Starting scheduler on CPU0
I (431) main_task: Started on CPU0
I (431) main_task: Calling app_main()
I (441) uart: queue free spaces: 10
[00:00:00.451] LOG -- btstack_run_loop_freertos.c.236: run loop init, task 0x40822f60, queue item size 8
[00:00:00.451] LOG -- main.c.283: transport_register_packet_handler
[00:00:00.451] LOG -- btstack_tlv.c.48: TLV Instance 0x42092000
[00:00:00.461] LOG -- le_device_db_tlv.c.159: num valid le device entries 0
Bluepad32 (C) 2016-2024 Ricardo Quesada and contributors.
Version: v4.1.0
BTstack: Copyright (C) 2017 BlueKitchen GmbH.
Platform: custom
custom: init()
Max connected gamepads: 4
BR/EDR support: disabled
BLE support: enabled
[00:00:00.481] LOG -- hci.c.5527: hci_power_control: 1, current mode 0
[00:00:00.491] LOG -- main.c.203: transport_init
[00:00:00.491] LOG -- main.c.214: transport_open
I (501) BLE_INIT: Using main XTAL as clock source
I (511) BLE_INIT: ble controller commit:[7491a85]
I (511) BLE_INIT: Bluetooth MAC: 40:4c:ca:40:86:ea
I (511) phy_init: phy_version 310,dde1ba9,Jun  4 2024,16:38:11
I (571) phy: libbtbb version: 04952fd, Jun  4 2024, 16:38:26
[00:00:00.571] LOG -- hci.c.8132: BTSTACK_EVENT_STATE 1
[00:00:00.581] CMD => 03 0C 00
Bluetooth Allowlist: Disabled
[00:00:00.581] EVT <= 0E 04 01 03 0C 00
```

ricardoquesada, 2024-09-18:
I tried with two different esp32c6, and an h2... similar results.

(BTW, esp32-p4, AFAIK, does not contain Bluetooth).

ricardoquesada, 2024-09-18:
and I compiled with latest esp-idf 5.3 (lastest commit from the 5.3 branch) with similar results.

mringwal, 2024-09-18:
Thanks for the logs! Well, you're not correct. The last line shows the HCI Command Complete Event for the HCI Reset we sent earlier, which, is probably a good sign. Looks like the sending flow control logic / resp. their assumptions are broken, and hence, should be straight forward to fix. I'll have a look (hopefully soon).

I also got a ESP32-P4 Dev Kit. The ESP32-P4 does not have Bluetooth, but the dev kit comes with another ESP32-C6 on board and ESP-IDF seems to provide an abstraction (similar to our hci_transport_t) to talk to it. This isn't working out-of-the-box either, but I'll look at the ESP32-C6 first.

mringwal, 2024-09-30:
Hi @ricardoquesada Found a calm evening to look into this. 

On the C6 with ESP-IDF v5.3, the '`notify_host_send_available` callback of the VHCI controller does not seem to get called. BTstack waits for this, before sending the next HCI packet.

I first thought that this callback is not used in NimBLE, but now I can see that they protect the send operation with a semaphore which is only given by this callback. Weird. Either, there's a difference in the initialization, or.. don't know.

Anyway, I've added a quick fix that assumes that you can send a new packet when the old one is sent, which seems to work. Still need to either find an actual bug to report to Espressif...

mringwal, 2024-10-01:
https://github.com/espressif/esp-idf/issues/14655

ricardoquesada, 2024-10-01:
Thanks for looking into it. I'll test it tonight.

ricardoquesada, 2024-10-02:
@mringwal  I've just tried latest `develop` btstack branch, and I'm still triggering the bug. I guess the patch didn't land in `develop` yet, is that correct? ty!

mringwal, 2024-10-02:
Ups. Sorry. The workaround/hack is on develop-esp32-c6

(Again, it's a quick fix just to see that it's working in general, but cannot be merged until Espressif either tells us that we're using their API wrong or fix the implementation on C6/H2).

Nuts-Underline, 2024-11-28:
new ESP32 like C6 having diferent realisation of esp_vhci_host_send_packet
esp32/с3/s3 callng `API_vhci_host_send_packet` from _libbtdm_
esp32c2/c5/c6/h2 - `r_ble_hci_trans_hs_cmd_tx` and `r_ble_hci_trans_hs_acl_tx` from _libble_  (derived from mynewt-nimble project)

mringwal, 2024-12-10:
Hi there. I've updated the code on the develop branch to use synchronous api on newer Controllers, plus fixed a function rename in esp-idf v5.3. Please let me know if this works for you. (Just tested successfully on ESP32-C6)

closed 2024-12-10 by mringwal

ricardoquesada, 2024-12-19:
great. ty. I'll set it tonight

ricardoquesada, 2024-12-20:
tested on esp32-c6. 
I was able to connect a device  using it.

thanks for the fix.

ricardoquesada, 2024-12-21:
confirm that it works Ok on esp32-h2 as well