bluekitchen/btstack #633: HID_SUBEVENT_DESCRIPTOR_AVAILABLE fires only for the first connection

Slion, 2024-10-14:
**Describe the bug**

I'm trying to get HID host to work with multiple connections but `HID_SUBEVENT_DESCRIPTOR_AVAILABLE` never fires past the first connection. Logs attached.

**To Reproduce**

Try connect 2 HID device to your host.

**Expected behavior**

`HID_SUBEVENT_DESCRIPTOR_AVAILABLE` should fire for second device connection too.

**HCI Packet Logs**
[HCI-Desc-Not-Available.zip](https://github.com/user-attachments/files/17359144/HCI-Desc-Not-Available.zip)

**Environment:  (please complete the following information):**
 - Current BTstack branch: develop
 - Bluetooth Controller: Pico W
 - Remote device: Android HID client

mringwal, 2024-10-14:
Please check the number of L2CAP Channels (MAX_NR_L2CAP_CHANNELS) unless malloc is used and/or increase by one.
It looks like the SDP Query fails as there are no L2CAP Channels available.

Slion, 2024-10-14:
Following your answer on the mailing I did gave it a try with MAX_NR_L2CAP_CHANNELS 16 and MAX_NR_L2CAP_SERVICES 16 but it did not do any better.

closed 2024-10-14 by Slion

Slion, 2024-10-15:
I had possibly two issues going on here, one I guess with the limited L2CAP channels that did not give any error back.
The other one has to do with the buffer you pass to `hid_host_init`. It needs to be large enough to contain all the HID report descriptor for all the devices you are connecting. When space runs out in that buffer `HID_SUBEVENT_DESCRIPTOR_AVAILABLE` is fired with with status `ERROR_CODE_MEMORY_CAPACITY_EXCEEDED` as returned by `hid_subevent_descriptor_available_get_status`.

Slion, 2024-10-16:
A quote from @mringwal from the [mailing list](https://groups.google.com/g/btstack-dev/c/nS6fYwp9_OY): 

> MAX_NR_HID_HOST_CONNECTIONS does not match MAX_NR_HCI_CONNECTIONS, but it would make sense if HID is your main application. You also need 2 L2CAP Connections for each HID Device.

I find HCI connections are rather expensive each takes almost 4KB. Lucky the L2CAP channels are cheap, I get 16 for around 1KB.