bluekitchen/btstack #638: Error Raspi make

Flouzl, 2024-10-26:
**Describe the bug**

I compiled the btstack/port/raspi file and it gave me a function error. I am on Raspberry pi 3 b

**To Reproduce**
1. Run 'git clone https://github.com/bluekitchen/btstack.git'
2. Run 'cd btstack/port/raspi'
3. Run 'make'
4. after that we have the error

**Expected behavior**

I would like to compile the file to use the library

**Environment: **
 - Current BTstack branch: [Raspberry pi 3 b]

**Additional context**
log :

```

flouzi@raspberrypi:~ $ cd btstack/port/raspi
flouzi@raspberrypi:~/btstack/port/raspi $ make
arm-linux-gnueabihf-gcc -I. -I../../src/ble -I../../src/classic -I../../src/mesh -I../../src -I../../3rd-party/bluedroid/decoder/include -I../../3rd-party/bluedroid/encoder/include -I../../3rd-party/hxcmod-player -I../../3rd-party/lwip/core/src/include/ -I../../3rd-party/lwip/dhcp-server -I../../3rd-party/md5 -I../../3rd-party/micro-ecc -I../../3rd-party/yxml -I../../platform/lwip -I../../platform/lwip/port -g -Wall -Werror -I../../platform/embedded -I../../platform/posix -I../../chipset/bcm -I../../3rd-party/tinydir -I../../3rd-party/rijndael   -c -o btstack_chipset_bcm.o ../../chipset/bcm/btstack_chipset_bcm.c
arm-linux-gnueabihf-gcc -I. -I../../src/ble -I../../src/classic -I../../src/mesh -I../../src -I../../3rd-party/bluedroid/decoder/include -I../../3rd-party/bluedroid/encoder/include -I../../3rd-party/hxcmod-player -I../../3rd-party/lwip/core/src/include/ -I../../3rd-party/lwip/dhcp-server -I../../3rd-party/md5 -I../../3rd-party/micro-ecc -I../../3rd-party/yxml -I../../platform/lwip -I../../platform/lwip/port -g -Wall -Werror -I../../platform/embedded -I../../platform/posix -I../../chipset/bcm -I../../3rd-party/tinydir -I../../3rd-party/rijndael   -c -o btstack_chipset_bcm_download_firmware.o ../../chipset/bcm/btstack_chipset_bcm_download_firmware.c
../../chipset/bcm/btstack_chipset_bcm_download_firmware.c:177:13: error: ‘bcm_send_hci_baudrate’ defined but not used [-Werror=unused-function]
  177 | static void bcm_send_hci_baudrate(void){
      |             ^~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
make: *** [<commande interne> : btstack_chipset_bcm_download_firmware.o] Erreur 1
flouzi@raspberrypi:~/btstack/port/raspi $ ^C
flouzi@raspberrypi:~/btstack/port/raspi $ 
```

Flouzl, 2024-10-29:
thank you I can complie btstack but the problem is I can’t make 
cd example 
make ./le_counter 

is doesn't work

mringwal, 2024-10-29:
please follow port/raspi/README.md

after running make in port/raspi, all examples are built in that folder, so you can run e.g. gatt_counter directly.
(there's no le_counter anymore, did you read that name in the documentation somewhere?)

closed 2024-10-29 by mringwal