bluekitchen/btstack #640: have a way to detect BTstack version at compile time

ricardoquesada, 2024-10-31:
**Is your feature request related to a problem? Please describe.**
Recently the API to use the `btstack_hid_parser_t` changed in a way that is not backward compatible.
In added `usage_iterator`.

Libraries that depend on BTstack (like Bluepad32) might use latest BTstack version (when bundled with Bluepad32), or might use the BTstack bundled with PicoSDK (which uses the old API).

For example,  Bluepad32 (and other libraries) could use something like:

```c
#if defined __has_include
#  if __has_include (<btstack_version.h>)
#    include <btstack_version.h>
#    ... compare-version
#    ... define use_new_api..
#  else
#   ... define use_old_api...
#endif
```

**Describe the solution you'd like**
Before releasing a new stable version, it would be great to have a header file that has the BTstack version defined.
Based on that I can either use one API, or the other.

**Describe alternatives you've considered**

**Additional context**
Add any other context or screenshots about the feature request here.

mringwal, 2024-10-31:
Hi @ricardoquesada Fair request. There's already another feature request on the issue tracker.

Ideally, it would use the version and git hash, however, that's not possible across any all platforms. 

However, updating a version file manually per release sounds doable.

mringwal, 2024-12-20:
We've added a btstack_version.h on develop branch file that's included by btstack_defines.h to simplify the transition.
If BTSTACK_VERSION_MAJOR is not defined, it's v1.6.1 or older. Otherwise you can check the version.

closed 2024-12-20 by mringwal