bluekitchen/btstack #641: dont work running example

Flouzl, 2024-11-03:
**Describe the bug**

hi,
I want to use an example on my raspiberry b but it doesn't work

**To Reproduce**

Steps to reproduce the behavior:
1. Run  ' cd btstack/port/raspi'
2. Start action './gatt_counter'

**Logs**

Packet Log: /tmp/hc1_dump.pklg
H5, 921600, BT_REG_EN at GPIO 128, Reset Controller Phase 1: Download firmware Download firmware failed

mringwal, 2024-11-04:
Which Raspberry Pi do you have?
Please try after a power cycle.

Please post hci_dump.pklg (you need to .zip it to for upload)

Flouzl, 2024-11-06:
hi i on raspberry pi 3b 
what is a power cycle ?

[hci_dump.zip](https://github.com/user-attachments/files/17651342/hci_dump.zip)

mringwal, 2024-11-06:
Power Cycle = turn power off, wait, turn it on again.

Roy in 'The IT Crowd': "Have you tried turning it off and on again".
https://www.youtube.com/watch?v=5UT8RkSmN4k

mringwal, 2024-11-06:
The log says "Unable to open port /dev/serial1"
Does /dev/serial1 exist on your installation and do you you have the necessary rights to open it?

Did you disable bluez as described in the readme?

Flouzl, 2024-11-09:
Yes I disble bluez but is dont work and when I import 'btstack.h' the Library not exists

Flouzl, 2025-01-10:
it is possible to make vocal on Discord for talk about btstack: my user name is Flouzi

mringwal, 2025-01-12:
Hi. I'd prefer to answer here.

Please explain in detail what you've tried, what happened, and what you've expected.

Flouzl, 2025-01-21:
I installed the library on my raspberry pi but it is not recognized when I include in file in c #include btstack.h and I wanted to do this again and I can't find where is the code to receive the cover and display on an LCD screen https://bluekitchen-gmbh.com/avrcp-cover-art-in-ios-android/

mringwal, 2025-01-22:
Hi there. Thanks for the link to our blog. 

In the Cover Art blog post, we were using a Raspberry Pi Pico W, which is a microcontroller with two ARM Cortex M0+ Cores. It's much smaller than your Raspberry Pi 3b. We don't have an example that demonstrates AVRCP Cover Art on a Raspberry Pi with regular display.

To run BTstack, please follow the readme in btstack/port/raspi. You can either cross compile on a desktop in a Docker container or with a cross-compile toolchain. Or, you can compile the BTstack examples on the Raspberry Pi itself. You can try the a2dp_sink_demo which allows to setup an AVRCP Cover Art connection with 'd', then press 'j' to get info on the current song, and finally, press '@' to download the current image as cover.jpg into the current folder.

Flouzl, 2025-01-27:
I have the same error, even if I recompiled all btstack with docker

Packet Log: /tmp/hc1_dump.pklg
H5, 921600, BT_REG_EN at GPIO 128, Reset Controller Phase 1: Download firmware Download firmware failed

mringwal, 2025-01-29:
Well. Before it looked like a compile error " I include in file in c #include btstack.h..". 
Now, you managed to compile it somehow but you get the error that firmware download failed.
Can you send another packet log?

Flouzl, 2025-02-09:
In the first picture are the folder were he compiled for raspi and this is the log [image]

```Flouzi@raspberrypi: S sudo systemctl stop hciuart
flouzi@raspberrypi: - S sudo systemctl stop bthelper
Failed to stop bthelper.service: Unit bthelper.service not loaded. flouzi@raspberrypi: $ sudo systemctl stop bluetooth flouzi@raspberrypi: $ cd btstack/port/raspi
- btstaclflouzi@raspberrypi:~/btstack/port/raspi.$ ./azdp_sink_demo
Packet Log: / tmp/hc1_dump. pklg
к_кеу-н5, 921600, BT_REG_EN at GPIO 128, Reset Controller
OPhase 1: Download firmware
Download firmware failed ```

mringwal, 2025-02-09:
Please post hci_dump.pklg again (you need to .zip it to for upload)

mringwal, 2025-02-16:
Hi. From the hci_dump, it looks like the code expects the serial port to be /dev/tty.serial1, but the open fails.
In the code documentation, there's a note that Model 3A and 3B have not been tested by us.

As an alternative: could you try to run the new port/linux? It tries to access hci0 the same way as BlueZ.

Flouzl, 2025-02-26:
There dont have port/linux

mringwal, 2025-02-26:
Please check if you're on the develop branch and have the latest version:

https://github.com/bluekitchen/btstack/tree/develop/port/linux

Flouzl, 2025-02-26:
I get this message ans i can Connect my phone to the Bluetooth 

```flouzi@raspberrypi:~/btstack/port/linux/build $ ./a2dp_sink_demo
Packet Log: /tmp/hci_dump_a2dp_sink_demo.pklg
Bluetooth device: hcio
No audio playback.
Audio will be stored to 'a2dp_sink_demo.wav' file.
Starting BTstack ...
Failed to bind HCI socket: Operation not permitted ```