bluekitchen/btstack #644: STATIC_ASSERT compilling issue in develop branch

Nuts-Underline, 2024-11-28:
switched to develop branch and trying to compile _nrf5-cinnamon_ port
```
........../nRF5_SDK_17.1.0_ddde560/components/libraries/util/app_util.h:256: error: "STATIC_ASSERT" redefined [-Werror]
  256 | #define STATIC_ASSERT(...)                                                                          \
      | 
In file included from ../../port/radio_nrf5.c:41:
../../../../src/btstack_debug.h:92: note: this is the location of the previous definition
   92 | #define STATIC_ASSERT(COND,MSG) typedef char static_assertion_##MSG[(COND)?1:-1]
      | 
cc1: all warnings being treated as errors

```
as i can see _btstack_debug.h_ having a lot of changes, so its conflicting with _nRF5_SDK_ now

mringwal, 2024-12-02:
hi there. we've been busy completing our new Bluetooth listing. I'll first need to integrate some branches created in parallel before going back to this. We'll probably drop STATIC_ASSERT again (and replace it's usage in our code by simpler checks)

mringwal, 2025-01-30:
STATIC_ASSERT has been removed completely as it's tricky to define in a portable way (Nordic's nRF only needs to make it work on gcc for example).

closed 2025-01-30 by mringwal