bluekitchen/btstack #645: Method of isolating (pico-w) example files to separate clean and tidy project.

ladmanj, 2024-12-03:
Hello, 
please forgive me a bit off-topic question (maybe not?).

I would like to start a separate project based on a2dp_audio_sink example.c
Whatever i tried i can't setup my CMakeLists.txt to point to the dependencies from btstack.
I am bluestack beginner and also cmake ignorant. 
I need to spend a while reading cmake documentation, but in the mean time I would appreciate an example how to do it in simple steps.
I was searching the internet for a manual or  someones blog or video guide.

What I found is that everyone is struggling in the same way as me. Everyone ended with modifying the example code in the directories where vscode pico extension fetched it. Even if that mean modifying files in windows program files directory as an administrator !!!

It would be great if you kindly document the right way doing it.

Thank you very much
Jakub

mringwal, 2024-12-03:
Hi. This is probably more related to the Pico SDK build system than to BTstack. 

As far as I know, you should be able to follow the Raspberry Pi Getting Started tutorial on how to setup an out-of-of tree build / project.

In the past, I've worked on this https://github.com/mringwal/galactic-bluetooth-audio and just started a new Bluetooth Pico project (which won't actually use BTstack) here: https://github.com/bluekitchen/hci-usb-to-uart

I'm not familiar with VSCode and finally managed to configure CLion for my Pico projects. I suggest to first get it working in the shell before adding another layer of complexity.

Hope that helps.

closed 2024-12-03 by mringwal

ladmanj, 2024-12-03:
I did start from the raspberry pico getting started tutorial and it works great, until i tried to integrate the btstack.

I don't know how to point the target project to the btstack library.

I believe that's not limited to vscode or pico in particular. It is related to how to link to the btstack libs using cmake.

Thank you
J.

mringwal, 2024-12-04:
I don't understand why you need to point the target project to the btstack library. As far as I know, the Pico SDK contains a copy of BTstack as a submodule. Could you check if BTstack is at pico-sdk/lib/btstack (and contains files)?

ladmanj, 2024-12-04:
this is what was missing in my CMakeLists.txt and I was unable to find it out:
```
set(BTSTACK_ROOT ${PICO_SDK_PATH}/lib/btstack)
set(BTSTACK_EXAMPLE_PATH ${BTSTACK_ROOT}/example)
set(BTSTACK_3RD_PARTY_PATH ${BTSTACK_ROOT}/3rd-party)

set(BTSTACK_CONFIG_PATH ${CMAKE_CURRENT_LIST_DIR})

target_link_libraries( blah blah
...
        pico_btstack_classic
        pico_btstack_ble
        pico_btstack_sbc_decoder    <- this one
)
```

The project you linked helped me a lot!

Thank you
J.