bluekitchen/btstack #646: Determine whether bnep_connect failed if the remote device doesn't support the desired UUID.

western-hoolock, 2024-12-08:
I'm using BTStack 1.6.1 to establish a BNEP connection by calling `bnep_connect` for the NAP remote UUID to use the remote device's hotspot functionality.
This works fine when the device actually has hotspot functionality enabled but when it doesn't there is no obvious event that is emitted to signal that this was the case?

Only notable log is generated here:
https://github.com/bluekitchen/btstack/blob/2b49e57bd1fae85ac32ac1f41cdb7c794de335f6/src/classic/bnep.c#L942
Which shows the BNEP channel failed because of reason `1` which is `BNEP_RESP_SETUP_INVALID_DEST_UUID`, so it properly detected it, but since the BNEP channel isn't in the open state yet nothing is actually emitted.

What's the recommended way to detect that this failed?

mringwal, 2024-12-09:
Thanks for reporting this. You're right, if there's an error in the BNEP setup, the application should receive a BNEP_EVENT_CHANNEL_OPENED event with some error code.  We'll have a look at it.

As a quick fix, you can call bnep_emit_open_channel_complete(channel, ERROR_CODE_UNSPECIFIED_ERROR). It looks like we could/should add a custom BNEP_... errror code as well.

western-hoolock, 2024-12-09:
> Please try the current version on develop, fe3c1d and let me know if this works for you. I've moved the bnep setup connection response codes to bnep.h

Just tried a couple of different scenarios including the `BNEP_SETUP_CONNECTION_RESPONSE_INVALID_DEST_UUID` one, seems to work fine now!
Thanks for the quick fix.

mringwal, 2024-12-10:
Thanks for the validation. We plan to make a new release before Christmas.

closed 2024-12-10 by mringwal