bluekitchen/btstack #648: aw859a module support

Nuts-Underline, 2024-12-19:
Trying to run posix-h4 examples on orangepi zero 2w. Wireless module aw859a - WiFi and BT. BT part connected to ttyBT0 as H4 device. (hciattach and other bluez stuff working, then disabled in OS). But no proper reaction from BTstack.
for example
```
_/btstack-develop/port/posix-h4$_ ./gap_le_advertisements 
Packet Log: /tmp/hci_dump.pklg
H4 device: /dev/ttyBT0
Local version information:
- HCI Version    0x0009
- HCI Revision   0x0400
- LMP Version    0x0009
- LMP Subversion 0x0400
- Manufacturer 0x0000
Unknown manufacturer / manufacturer not supported yet.
Local name: SPREADTRUM BT
BTstack up and running on 27:93:31:14:22:11.
TLV path: /tmp/btstack_27-93-31-14-22-11.tlv
```
no other rection, but i have many BLE adv. sources around

So, this BT chip not supported yet, OK. 
Question is: _Does it mean that unsupported H4 chip have no any default init to work somehow?_ (seems like bluez init it on 115200 baud)

mringwal, 2024-12-19:
The message could be improved. It only means that there's no special support for it. It should work in general. Please post the packet log at /tmp/hci_dump.pklg to see why there are no advertisements.

Nuts-Underline, 2024-12-19:
here
[hci_dump.zip](https://github.com/user-attachments/files/18198305/hci_dump.zip)

Nuts-Underline, 2024-12-19:
looks like its using patched _hciattach_opi_ with closed sources :(

mringwal, 2024-12-19:
Depending on your motivation, the simplest approach would be to use the orangepi with an USB Bluetooth dongle that BTstack supports.

If you want to work on adding support for AllWinner Controllers, we're happy to help, I just didn't encounter one before and we'd need both some kind of documentation + a way to get the firmware patches.

Nuts-Underline, 2024-12-20:
Well, i just want to check BT 5.xx options (PHYs, ext. adv. so on) of this adapter in H4 mode. As hciattach_opi is closed (SPREADTRUM NDA i think) i cant add this adapter to btstack.
Qick and durty solution for orange pi's is run and stop _aw859a-bluetooth.service_  (sprd_bluetooth.service) only **once**, then btstack examples working OK.

_/usr/bin/hciattach_opi -s 1500000 /dev/ttyBT0 sprd_

log with some details 

```
Dec 20 11:01:58 orangepizero2w hciattach_opi[1351]: SPRD Bluetooth init uart with init speed:1500000, final_speed:1500000, type:HCI UART H4
Dec 20 11:01:58 orangepizero2w hciattach_opi[1351]: SPRD Bluetooth: Attempt to load conf from /lib/firmware/bt_configure_pskey.ini
Dec 20 11:01:58 orangepizero2w hciattach_opi[1351]: SPRD Bluetooth: device_class ->  0x001F00
Dec 20 11:01:58 orangepizero2w hciattach_opi[1351]: SPRD Bluetooth: feature_set ->  0xBF, 0xFF, 0x8D, 0xFE, 0xDB, 0x3D, 0x7B, 0x87, 0xFF, 0xA7, 0xFF, 0x7F, 0x00, 0xE0, >
Dec 20 11:01:58 orangepizero2w hciattach_opi[1351]: SPRD Bluetooth: device_addr ->  0x6D, 0x6D, 0x8E, 0x8E, 0x8F, 0x8F
Dec 20 11:01:58 orangepizero2w kernel: mtty_open device success!
Dec 20 11:01:58 orangepizero2w hciattach_opi[1351]: SPRD Bluetooth: comp_id ->  0x01EC
Dec 20 11:01:58 orangepizero2w hciattach_opi[1351]: SPRD Bluetooth: g_sys_uart0_communication_supported ->  1
Dec 20 11:01:58 orangepizero2w hciattach_opi[1351]: SPRD Bluetooth: cp2_log_mode ->  1
Dec 20 11:01:58 orangepizero2w hciattach_opi[1351]: SPRD Bluetooth: LogLevel ->  0xFF
Dec 20 11:01:58 orangepizero2w hciattach_opi[1351]: SPRD Bluetooth: g_central_or_perpheral ->  0
Dec 20 11:01:58 orangepizero2w hciattach_opi[1351]: SPRD Bluetooth: Log_BitMask ->  0xFFFF
Dec 20 11:01:58 orangepizero2w hciattach_opi[1351]: SPRD Bluetooth: super_ssp_enable ->  0
Dec 20 11:01:58 orangepizero2w hciattach_opi[1351]: SPRD Bluetooth: common_rfu_b3 ->  0
Dec 20 11:01:58 orangepizero2w hciattach_opi[1351]: SPRD Bluetooth: common_rfu_w ->  0x00000000, 0x00000000
Dec 20 11:01:58 orangepizero2w hciattach_opi[1351]: SPRD Bluetooth: le_rfu_w ->  0x00000000, 0x000000000
Dec 20 11:01:58 orangepizero2w hciattach_opi[1351]: SPRD Bluetooth: lmp_rfu_w ->  0x00000000, 0x000000000
Dec 20 11:01:58 orangepizero2w hciattach_opi[1351]: SPRD Bluetooth: lc_rfu_w ->  0x00000000, 0x000000000
Dec 20 11:01:58 orangepizero2w hciattach_opi[1351]: SPRD Bluetooth: g_wbs_nv_117 ->  0x004D
Dec 20 11:01:58 orangepizero2w hciattach_opi[1351]: SPRD Bluetooth: g_wbs_nv_118 ->  0x0166
Dec 20 11:01:58 orangepizero2w hciattach_opi[1351]: SPRD Bluetooth: g_nbv_nv_117 ->  0x004D
Dec 20 11:01:58 orangepizero2w hciattach_opi[1351]: SPRD Bluetooth: g_nbv_nv_118 ->  0x0166
Dec 20 11:01:58 orangepizero2w hciattach_opi[1351]: SPRD Bluetooth: g_sys_sco_transmit_mode ->  0
Dec 20 11:01:58 orangepizero2w hciattach_opi[1351]: SPRD Bluetooth: audio_rfu_b1 ->  0
Dec 20 11:01:58 orangepizero2w hciattach_opi[1351]: SPRD Bluetooth: audio_rfu_b2 ->  0
Dec 20 11:01:58 orangepizero2w hciattach_opi[1351]: SPRD Bluetooth: audio_rfu_b3 ->  0
Dec 20 11:01:58 orangepizero2w hciattach_opi[1351]: SPRD Bluetooth: audio_rfu_w ->  0x00000000, 0x00000000
Dec 20 11:01:58 orangepizero2w hciattach_opi[1351]: SPRD Bluetooth: g_sys_sleep_in_standby_supported ->  1
Dec 20 11:01:58 orangepizero2w hciattach_opi[1351]: SPRD Bluetooth: g_sys_sleep_master_supported ->  1
Dec 20 11:01:58 orangepizero2w hciattach_opi[1351]: SPRD Bluetooth: g_sys_sleep_slave_supported ->  1
Dec 20 11:01:58 orangepizero2w hciattach_opi[1351]: SPRD Bluetooth: power_rfu_b1 ->  0
Dec 20 11:01:58 orangepizero2w hciattach_opi[1351]: SPRD Bluetooth: power_rfu_w ->  0x00000000, 0x00000000
Dec 20 11:01:58 orangepizero2w hciattach_opi[1351]: SPRD Bluetooth: win_ext ->  40
Dec 20 11:01:58 orangepizero2w hciattach_opi[1351]: SPRD Bluetooth: edr_tx_edr_delay ->  6
Dec 20 11:01:58 orangepizero2w hciattach_opi[1351]: SPRD Bluetooth: edr_rx_edr_delay ->  8
Dec 20 11:01:58 orangepizero2w hciattach_opi[1351]: SPRD Bluetooth: tx_delay ->  12
Dec 20 11:01:58 orangepizero2w hciattach_opi[1351]: SPRD Bluetooth: rx_delay ->  34
Dec 20 11:01:58 orangepizero2w hciattach_opi[1351]: SPRD Bluetooth: bb_rfu_w ->  0x00000000, 0x00000000
Dec 20 11:01:58 orangepizero2w hciattach_opi[1351]: SPRD Bluetooth: agc_mode ->  0
Dec 20 11:01:58 orangepizero2w hciattach_opi[1351]: SPRD Bluetooth: diff_or_eq ->  0xFF
Dec 20 11:01:58 orangepizero2w hciattach_opi[1351]: SPRD Bluetooth: ramp_mode ->  0
Dec 20 11:01:58 orangepizero2w hciattach_opi[1351]: SPRD Bluetooth: modem_rfu_b1 ->  0
Dec 20 11:01:58 orangepizero2w hciattach_opi[1351]: SPRD Bluetooth: modem_rfu_w ->  0x00000000, 0x00000000
Dec 20 11:01:58 orangepizero2w hciattach_opi[1351]: SPRD Bluetooth: yujian.qin marlin3_rf_preload
Dec 20 11:01:58 orangepizero2w hciattach_opi[1351]: SPRD Bluetooth: sprd_vnd_send_hci_vsc
Dec 20 11:01:58 orangepizero2w hciattach_opi[1351]: SPRD Bluetooth: a2 fc fc 00 e0 00 e0 00 e0 00 e0 00 e0 00 e0 15
Dec 20 11:01:58 orangepizero2w hciattach_opi[1351]: SPRD Bluetooth: 41 15 3a 15 34 15 2e 15 27 15 21 15 17 15 11 15
Dec 20 11:01:58 orangepizero2w hciattach_opi[1351]: SPRD Bluetooth: 0b 15 07 15 3b 15 37 15 33 15 2f 15 2b 15 27 15
Dec 20 11:01:58 orangepizero2w hciattach_opi[1351]: SPRD Bluetooth: 23 15 1f 15 1b 15 17 15 13 15 0f 15 0b 15 08 15
Dec 20 11:01:58 orangepizero2w hciattach_opi[1351]: SPRD Bluetooth: 04 15 00 15 06 15 06 15 06 15 06 15 06 15 06 15
[... cut ...]
Dec 20 11:01:58 orangepizero2w hciattach_opi[1351]: SPRD Bluetooth: Received event, len: 9
Dec 20 11:01:58 orangepizero2w hciattach_opi[1351]: SPRD Bluetooth: 04 0e 06 01 a1 fc 00 00 01
Dec 20 11:01:58 orangepizero2w hciattach_opi[1351]: SPRD Bluetooth: hw_core_cback hw_core_cback response: [0xFCA1, 0x0000, 0x01]
Dec 20 11:01:59 orangepizero2w hciattach_opi[1351]: SPRD Bluetooth post process
Dec 20 11:01:59 orangepizero2w hciattach_opi[1351]: SPRD Bluetooth: Done setting line discpline
Dec 20 11:01:59 orangepizero2w hciattach_opi[1351]: Device setup complete

```

Nuts-Underline, 2024-12-20:
my question is answered, so closing issue

closed 2024-12-20 by Nuts-Underline

mringwal, 2024-12-20:
Glad you've found a working setup. If needed, one could start recording the UART communication with the Bluetooth Controller somehow and compare that to the firmware blobs on the Orange Pi. To test new Bluetooth LE features (with BTstack), I'd recommend to use a Nordic nRF5x Controller with the HCI UART example. The latest nRF54L15 DK already has experimental support for Channel Sounding, while the nRF52840 USB dongle supports at least new PHYs, Ext Adv, etc. and is cost effective.