bluekitchen/btstack #649: develop branch: pico-w not linking. missing hci_event_builder.c

ricardoquesada, 2024-12-19:
**Describe the bug**

For Pico-W:

- `master` branch: compiles and links Ok.
- `develop` branch: compiles Ok, but does not link. The file `hci_event_builder.c` is not being linked.

**To Reproduce**

Probably my understanding is that this is Working-as-intended. For example, the test "test/gatt-service-client/CMakeLists.txt" includes that files manually.

But the different ports (stm32, esp32, etc...) have them included by default in the `Makefile` by default.

**Expected behavior**

Some Pico-w projects based on btstack 1.6.1 will fail when migrated to latest develop btstack branch.
They will need to manually include `hci_event_builder.c` in their `CMakeLists.txt`

The expected behavior is that `hci_event_builder.c` should be included automatically in PicoW.

**Environment:  (please complete the following information):**
 - Current BTstack branch: develop
 - Bluetooth Controller: Pico W
 - Remote device: N/A

**Additional context**
Add any other context about the problem here.

peterharperuk, 2024-12-19:
The pico-sdk has cmake files for btstack, so you should probably raise this as an issue here https://github.com/raspberrypi/pico-sdk/
We have discussed how to address build issues between btstack releases but I don't think we came to a conclusion. At the least we need an accessible version number I guess.

ricardoquesada, 2024-12-19:
ah... got it.
That's why I couldn't find the CMakeLists.txt file in btstack.
It is here: https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2_common/pico_btstack/CMakeLists.txt

Thinking aloud, perhaps btstack should have a `CMakeLists.txt` with the "core" features, and then each port includes the rest (??).

In any case, thanks. I've just filed this bug on pico-sdk: https://github.com/raspberrypi/pico-sdk/issues/2142

mringwal, 2024-12-20:
Thanks for the suggestions. We've added the compile time version in #640.

As for the CMakeLists.txt. In most ports, we glob all C files, which deals with new files. Not sure if that's an option for Pico SDK.
We have an internal suggestion for a more standard CMake build files, placed in the main folder, but I'm not too happy with it.

It's to late for this release, but I think we should move into that direction and e.g. provide a CMakeLists.txt file that includes all files in a particular folder. Having that one include subsdirs, might be a bit tricky as one might not want to compile src/classic for LE-only targets (or vice-versa).

mringwal, 2025-01-30:
This has been fixed in pico-sdk develop branch.

closed 2025-01-30 by mringwal

ricardoquesada, 2025-01-30:
I think you closed the wrong issue (I don't have the permissions to re-open).

this has not been fixed yet on btstack/pico-sdk.... the one that has been fixed is the crash.