bluekitchen/btstack #650: gattservice_subevent_hid_report_get_report() returning corrupted data (?)

ricardoquesada, 2024-12-20:
**Describe the bug**

`gattservice_subevent_hid_report_get_report()`  from `develop` branch is returning corrupted(?) data. Data that `btstack_hid_parser_` fails to parse correctly.

The problems seems to come from the packet that is received from `hids_client_packet_handler`.

It seems that `develop` branch reports more bytes (???).

full packet from develop branch:
`F2 18 19 01 00 00 01 11 00 01 1E 00 10 00 E5 84 AE 82 7A 80 30 7F 00 00 00 00 01 00 00 00`

full packet from main branch:
`F2 18 14 01 00 00 01 11 00 01 0D 86 45 83 8A 80 4F 7F 00 00 00 00 01 00 00 00`

As you can the packet from `develop` is longer... and it shouldn't be (???).

Attache

**To Reproduce**

Not sure how to reproduce it... this is part of the Bluepad32 example, and requires the entire Bluepad32 stack.
I can try to create a test case if needed... but I guess this should be reproducible with any BLE connection using `hids_client`.

**Expected behavior**

Packet should not be corrupted (?).

I can reproduce it %100 of the time.

Tested with btstack develop branch (latest as of today). Hash: 4da9eab982db614e33cc14c37d0c28d337f6cbd1

**HCI Packet Logs**
Attached both main and develop packages.
[hci_dump_develop.zip](https://github.com/user-attachments/files/18206752/hci_dump_develop.zip)
[hci_dump_main.zip](https://github.com/user-attachments/files/18206753/hci_dump_main.zip)

mringwal, 2024-12-20:
Thanks for reporting this just in time for the new release. Please verify the fix on the develop branch. 
We found & fixed a similar/related bug in the ancs_client.

ricardoquesada, 2024-12-20:
yay. working  ok. thanks for the rapid fix.

closed 2024-12-20 by ricardoquesada