bluekitchen/btstack #652: ESP32 max ACL connection count and its effect on inquiry/remote name requests.

western-hoolock, 2024-12-30:
Using BTStack 1.6.1 on an ESP32-WROVER-IE with ESP-IDF 5.3.0.

I'm doing inquiry and remote name requests pretty much in the same way as shown in the `gap_inquiry.c` example.
i.e. Start an inquiry, wait for `GAP_EVENT_INQUIRY_COMPLETE` event, and then check if we have found devices without a name and start querying those one by one.

This works fine with the ESP-IDF `CONFIG_BTDM_CTRL_BR_EDR_MAX_ACL_CONN` configuration variable set to the default of 2, but when setting that to 1 (or taking one up by establishing an actual ACL connection for example) the remote name requests start to fail silently.
`gap_remote_name_request` will return 0 but the `HCI_EVENT_REMOTE_NAME_REQUEST_COMPLETE`  event is then never seen in the packet handler.
Setting up an actual connection (through BNEP for example) does correctly return a 0x09 'Connection Limit Exceeded' error.

My Bluetooth knowledge is somewhat limited but I was always under the impression that inquiry/remote name requests etc. work at the baseband level without having to set up an ACL connection so I'm somewhat confused why this seems to count towards the ACL connection limit? Is this an 'ESP32 only' thing?

mringwal, 2025-01-07:
Hi. 

If there's an active connection, HCI Remote Name Request will query the name over  this connection. If not, the  Controller will open a connection, request the name and close the connection. So, your observation makes perfect sense and this not an ESP32 only issue.

However, if the HCI Remote Name Request failes, e.g. because there are no free connections, then you won't get an HCI_EVENT_REMOTE_NAME_REQUEST_COMPLETE. In this case you should get an HCI Command Status Event with the error code. 

I think it would make sense here to simulate at HCI_EVENT_REMOTE_NAME_REQUEST_COMPLETE and return the error message to keep application logic simple. If you want to test, you can match HCI_EVENT_COMMAND_STATUS and check hci_event_command_status_get_command_opcode(packet) == HCI_OPCODE_HCI_REMOTE_NAME_REQUEST && hci_event_command_status_get_status(packet) != ERROR_CODE_SUCCESS

western-hoolock, 2025-01-08:
Ok thanks that clears things up regarding the Remote Name Request but I'm still not entirely sure why an Inquiry itself would also take up a connection.

Simulating a `HCI_EVENT_REMOTE_NAME_REQUEST_COMPLETE` would be nice. In my case I had a little state machine that expected it to always fire in order to advances and it started to hang.
I expected it to always fire because some other events follow the same logic but I also have a device that's bordering on being out of range and its name can't always be resolved, sometimes `HCI_EVENT_REMOTE_NAME_REQUEST_COMPLETE` would fire but it would just be all zeroes for the name. So I just assumed that meant it failed and I would always get a `HCI_EVENT_REMOTE_NAME_REQUEST_COMPLETE` regardless but it seems that's not the case.

Which brings me to my follow up question:
All devices I have here (iPhone, laptops etc.) that are nearby, always return their name without having to do a Remote Name Request. I also have the device much further away mentioned above that isn't always returned in an Inquiry, sometimes it is, but without a name, and other times with a name. When it doesn't return a name, doing a Remote Name Request sporadically returns the name, but not always.
It is my understanding that since Bluetooth 2.1 the EIR bit can be set in the FHS response which indicates that an extended inquiry response will follow immediately. I assume that if that extended response never arrives, it is simply reported as having no name?

I am now contemplating not using Remote Name Requests at all since the vast majority of devices will have the EIR bit set and supply their name anyway and the ones without a name are likely out of range and not reliable to connect to anyway so I just don't report those. Does that sound reasonable in your experience?

mringwal, 2025-01-13:
Hi. I've added the code to emit a HCI_EVENT_REMOTE_NAME_REQUEST_COMPLETE if the command fails in 7bb9c8d (develop).

Could you verify that you get the event in your setup? It's not easy to test this with a regular Controller that supports multiple parallel Classic connections.

I haven't looked into the EIR details, but it's quite plausible that you can get an inquiry response without EIR data if the device is further away/connection is weak.

mringwal, 2025-01-15:
Thanks for testing and the detailed explanation - always good to know!

closed 2025-01-30 by mringwal