bluekitchen/btstack #655: gap_drop_link_key_for_bd_addr needs a reboot to take effect

Slion, 2025-01-19:
**Describe the bug**

When deleting a classic pairing key using `gap_drop_link_key_for_bd_addr ` you need to reboot the board for it to take effect.
It looks like pairing key is deleted from storage but still persisted in memory.

**To Reproduce**

I've seen this happening with HID devices using HID host.

- Have the host pair with a keyboard.
- See that it works
- Delete pairing key and disconnect the device
- Device can still reconnect 
- Reboot the board
- Device can't reconnect

**Expected behavior**

Board reboot should not be needed.

**Environment:**
 - Current BTstack branch: develop
 - Bluetooth Controller: Pico W
 - Remote device: Logitech K810

**Additional context**
Add any other context about the problem here.

mringwal, 2025-01-21:
Thanks for reporting. In the pico-sdk, btstack_link_key_db_tlv is used, which doesn't have any cache and directly maps link key operations to the underlying btstack_tlv implementation. 

Are you sure the HCI connection was disconnected?

Could you provide an HCI log for this use case (with ENABLE_LOG_INFO but without ENABLE_LOG_DEBUG)?
If the link key is found after the gap_drop_link_key_for_bd_addr was called, this would indicate a problem with the TLV implementation. However, it's not clear why a reboot would help in this case.

Slion, 2025-01-23:
Cannot reproduce it now. Thanks for taking a look though.

closed 2025-01-23 by Slion

Slion, 2025-01-30:
Actually I twisted things a bit here I'm afraid, possibly jumping to conclusions.
The actual issue was that I was getting ̀ L2CAP_CONNECTION_RESPONSE_RESULT_REFUSED_SECURITY` from `HID_SUBEVENT_CONNECTION_OPENED` and as an ran into that issue before I deleted the pairing but I would still get that same error until I rebooted the device. So I thought it would be reproducible with the steps given above too. Turns out it was not.
Sorry no logs and I don't know how to reproduce that security issue. Maybe it happens when the keyboard was paired with another but the board is still paired with the keyboard.

mringwal, 2025-01-30:
Hi. The only place I'm aware of where the link key is stored besides the persistent store is in the hci connection itself. It's not so easy to see how it is used, but if you first connect and this fails, as the remote doesn't have a key and the call gap_drop_link_key_for_bd_addr and trigger another pairing, while still connected, this key would be used. I've pushed a new commit that also marks the link key stored in the hci connection as invalid.

Slion, 2025-01-30:
>  I've pushed a new commit that also marks the link key stored in the hci connection as invalid.

See: b298729501ae132532200c66fbfc458898c994d2

> The only place I'm aware of where the link key is stored besides the persistent store is in the hci connection itself.

That must be it then, maybe there are a few not so common workflows where this somehow gets in the way.