bluekitchen/btstack #658: No BNEP_EVENT_CHANNEL_OPENED event after bnep_connect call.

western-hoolock, 2025-02-05:
BTStack 1.6.2 running on an ESP32 and remote device is Windows 10 with a generic Realtek USB dongle.

My code tries a couple of BT addresses from a list of paired devices until it finds one to which it can successfully setup a BNEP connection.
It does this by calling `bnep_connect` and wait for a `BNEP_EVENT_CHANNEL_OPENED` event. However today I encountered a situation where the L2CAP channel seems to have been setup successfully but then is promptly closed by the remote side for some reason (not sure why that happened) but causes BTStack to never emit a `BNEP_EVENT_CHANNEL_OPENED` event.

Here's the partial log:
```
[00:00:08.428] LOG -- bnep.c.1628: BNEP_CONNECT addr 00:E0:4C:AF:BE:EF
<omitted for brevity>
[00:00:08.628] LOG -- l2cap.c.1141: L2CAP_EVENT_CHANNEL_OPENED status 0x0 addr 00:E0:4C:AF:BE:EF handle 0x81 psm 0xf local_cid 0x44 remote_cid 0x41 local_mtu 1691, remote_mtu 1691, flush_timeout 0
[00:00:08.648] LOG -- l2cap.c.1158: ERTM mode 0, fcs enabled 1
[00:00:08.648] LOG -- bnep.c.1289: L2CAP_EVENT_CHANNEL_OPENED for BLUETOOTH_PSM_BNEP, status 0
[00:00:08.658] LOG -- bnep.c.1313: L2CAP_EVENT_CHANNEL_OPENED: outgoing connection
[00:00:08.668] LOG -- bnep.c.727: bnep_max_frame_size_for_l2cap_mtu:  1691 -> 1676
[00:00:08.698] LOG -- l2cap.c.3419: L2CAP signaling handler code 6, state 12
[00:00:08.698] LOG -- l2cap.c.1200: L2CAP_EVENT_CHANNEL_CLOSED local_cid 0x44
[00:00:08.698] LOG -- bnep.c.1345: L2CAP_EVENT_CHANNEL_CLOSED cid 0x44, channel 0x3ffe46c4
[00:00:08.708] LOG -- bnep.c.1351: L2CAP_EVENT_CHANNEL_CLOSED state 3
[00:00:08.718] LOG -- l2cap.c.2515: free channel 0x3ffe47b0, local_cid 0x0044
[00:00:08.718] LOG -- l2cap.c.1419: l2cap_stop_rtx for local cid 0x44
[00:00:08.728] LOG -- l2cap.c.363: Stop Retransmission timer
[00:00:08.728] LOG -- l2cap.c.349: Stop Monitor timer
[00:00:11.858] LOG -- hci.c.1292: Connection closed: handle 0x81, 00:E0:4C:AF:BE:EF
[00:00:11.858] LOG -- hci.c.8176: BTSTACK_EVENT_NR_CONNECTIONS_CHANGED 0
```

The remote machine decides to close the L2CAP channel. The BNEP state is still `BNEP_CHANNEL_STATE_CLOSED` and since no actual errors have occurred `bnep_emit_open_channel_complete` is never called anywhere.

This seems to be an edge case/bug that could potentially be handled here?
https://github.com/bluekitchen/btstack/blob/565100c1cae73dc1c5f9b239953dbdd14ae283d3/src/classic/bnep.c#L1352-L1360

mringwal, 2025-02-05:
Thanks for reporting this. Indeed, there's no event if there's any kind of failure before the BNEP connect message are exchanged.

I've extended the code on the develop branch to emit BNEP_EVENT_CHANNEL_OPENED with ERROR_CODE_REMOTE_USER_TERMINATED_CONNECTION.

Please let me know if this works for you.

western-hoolock, 2025-02-06:
Thanks for the commit, can confirm this works!

Had to figure out how to reproduce this, I was just lucky to have logging enabled when this happened yesterday. I probably turned off Bluetooth on the Windows machine at the right time (or it was the buggy Windows driver that causes me multiple blue screens of death several times per week...)

Anyway, I ended up reproducing the error by taking another ESP32 to act as a hotspot but always immediately close the channel instead of entering BNEP connection negotiation.
This time I got the `BNEP_EVENT_CHANNEL_OPENED` with `ERROR_CODE_REMOTE_USER_TERMINATED_CONNECTION` on the other end as expected.

mringwal, 2025-02-06:
Thanks for taking the time to verify this.

closed 2025-02-06 by mringwal