bluekitchen/btstack #665: HFP AG demo / Samsung buds 2 issue

amorniroli, 2025-03-03:
**Describe the bug**

Trying the HFP AG demo example using samsung buds 2 as headset but cannot receive event about established connection.

**To Reproduce**

Steps to reproduce the behavior:
1. Run example hfp_ag_demo.c
2. Put Samsung Buds 2 in pairing mode
3. Press 'a' key to establish connection
4. Wait for HFP_SUBEVENT_SERVICE_LEVEL_CONNECTION_ESTABLISHED  routed

**Expected behavior**

HFP_SUBEVENT_SERVICE_LEVEL_CONNECTION_ESTABLISHED event is not routed.

If I wait some time and then press 'a' again, I see that connection remains in HFP_W4_RETRIEVE_GENERIC_STATUS_INDICATORS state.

[image]

**HCI Packet Logs**

[samsung_buds.pklg.txt](https://github.com/user-attachments/files/19054731/samsung_buds.pklg.txt)
[samsung_buds.txt](https://github.com/user-attachments/files/19054730/samsung_buds.txt)

**Environment:  (please complete the following information):**
 - Current BTstack branch: develop (c468ca4b35f9c46377031835024df7d0cf7ba5d5)
 - Bluetooth Controller: CYBT-343026-02 (CYW20706)
 - Remote device: Samsung Buds 2

**Additional context**

mringwal, 2025-03-03:
Hi @amorniroli 

It looks like you're using a synchronous HCI transport which caused AG messages to be sent multiple times, e.g. the OK for "AT+BAC=1,2,16", which then might have cause other issues later in the SLC.

I've patched the libusb transport to behave like a synchronous one and was able to find a number of places to fix, which is available on develop.

Pease update to develop branch and try again + send packet log.

amorniroli, 2025-03-04:
Hi @mringwal 

both solutions (pulling develop and using sync or old commit and using async) work as expected. Thank you

[sync_h4_1f885aa.pklg.txt](https://github.com/user-attachments/files/19067152/sync_h4_1f885aa.pklg.txt)
[sync_h4_1f885aa.txt](https://github.com/user-attachments/files/19067149/sync_h4_1f885aa.txt)
[async_h4_c468ca4.pklg.txt](https://github.com/user-attachments/files/19067150/async_h4_c468ca4.pklg.txt)
[async_h4_c468ca4.txt](https://github.com/user-attachments/files/19067151/async_h4_c468ca4.txt)