bluekitchen/btstack #667: Raspberry PI 4B support

Ciantic, 2025-03-24:
I'm trying to implement this, but since I have no expertise I got a bit stuck. 

So far I have figured out how to get the LOCAL NAME printed out just once, notice: **BCM4345C0** this means something is working somehow already.

```
sudo ./hid_mouse_demo
Packet Log: /tmp/hci_dump.pklg
Raspberry Pi UART type 3
H4, 3000000, BT_REG_EN at GPIO 129, Warm Boot
Power cycle 0
Flow control 1
Terminal device /dev/serial1
baud 13
current serial terminal parameter baudrate: 9600, flow control: None
Local name: BCM4345C0
```

Steps I did to get this far:

## 1. In config config.txt
```
[all]
dtoverlay=dwc2
dtoverlay=miniuart-bt
dtparam=krnbt=off
```
(Adding `dtoverlay=disable-bt` didn't help.)

## 2. `ENABLE_CONTROLLER_WARM_BOOT`

## 3. Added constant `MODEL_4B`

## 4. hard coded `raspi_get_bluetooth_uart_type` to return `UART_HARDWARE_FLOW`

mringwal, 2025-03-24:
If you can get the local name, that's already quite good. Could you upload /tmp/hci_dump.plkg please? You can open it with Wireshark for analysis.

Also: you can try the new port/linux which uses the linux kernel driver to talk to the Bluetooth Controller. (You might need to switch to develop branch)

mringwal, 2025-03-24:
Well, it's a binary format which you can open with Wireshark.

It shows that communication works in general, that it can find the PatchRAM file for the 4345. After sending the command to change the baudrate to 3000000, we get errors like

hci_transport_h4.c.263: hci_transport_h4: invalid packet type 0x84

This indicates that:
- 3 mbps is not supported on that UART, or,
- flow control is not working.

Please try to use 0 as main baud rate to skip the baud rate change. If that works, you can try increasing to 230400 and 921600 to see if that works.

Ciantic, 2025-03-24:
@mringwal 

WOW! It works!

```
sudo ./hid_mouse_demo
Packet Log: /tmp/hci_dump.pklg
Raspberry Pi UART type 3
H4, 0, BT_REG_EN at GPIO 129, Warm Boot
Power cycle 0
Flow control 1
Terminal device /dev/serial1
baud 13
current serial terminal parameter baudrate: 9600, flow control: None
Local name: BCM4345C0
BTstack up and running on B8:27:EB:A2:C9:1B.
HID Connected, control mouse using 'a','s',''d', 'w' keys for movement and 'l' and 'r' for buttons...www
```

I forced `transport_config.baudrate_main = 0;`

I would like to make PR, but I have to figure out how to make this cleanly so that it works with older models too.

Ciantic, 2025-03-24:
I created a draft PR https://github.com/bluekitchen/btstack/pull/668 if someone else stumbles here, it explains how I got this functioning.

I don't understand the old logic there, so I have (for now) circumvented it with just `if (model == MODEL_4B) {` statements in few places.