bluekitchen/btstack #671: btstack_base64_encoder.h using POSIX only variable type

simo-scar-cob, 2025-04-01:
**Describe the bug**

While compiling port WinUSB on develop branch, compiler raises error for variable type: ssize_t ; used in btstack_base64_encoder.h file.  
Which acutally I think is only for POSIX systems (UNIX, Linux, etc..). 
On Windows the equivalent I think is SSIZE_T , defined in #include <basetsd.h>. 
This fix works for me, but I think that a separation based on OS is needed. 

**To Reproduce**

Steps to reproduce the behavior:
1. open develop branch on Visual Studio 2022 Community
2. run compilation for windows-winusb port

**Expected behavior**
Compile success

**Environment:  (please complete the following information):**
 - Current BTstack branch: develop
 - Bluetooth Controller: CSR 8510

mringwal, 2025-04-01:
Thanks for reporting. We'll discuss internally, which way to go.

mringwal, 2025-04-02:
We've added a typedef of SSIZE_T in btstack_defines.h on develop branch.