bluekitchen/btstack #674: Connection rejected due to limited resources

amorniroli, 2025-04-08:
Hi,

not sure if it's a bug (very similar to https://github.com/bluekitchen/btstack/issues/175)

I have commercial headset (SENA20S) that it's refusing to open SCO due to limited resources.

Very sample setup:

- infineon CYBT343026 (CYW20706)
- hfp_ag_demo.c (I just put the headset in pairing mode, then press 'a' to connect and pair, wait some time and then press 'b' to establish SCO)
- ENABLE_HFP_HF_SAFE_SETTINGS enabled in btstack_config.h

Attached log.txt (serial debug) and wireshark.txt (rename to pklg).

Any idea? The SCO is opened correctly if used with smartphone and other product based on cc2564, but at the moment I can't record HCI trace for that

[log.txt](https://github.com/user-attachments/files/19651962/log.txt)

[wireshark.txt](https://github.com/user-attachments/files/19652000/wireshark.txt)

amorniroli, 2025-04-08:
Some updates: I recall encountering a similar issue with the same headset in the past (https://github.com/bluekitchen/btstack/issues/424).

Could something have changed or been inadvertently broken since then?

Tomorrow I'll try the old commit of #424 and give you a feedback

mringwal, 2025-04-09:
Hi Alessandro. 

The log looks similar, good catch:

Could you:
- try adding ERROR_CODE_CONNECTION_REJECTED_DUE_TO_LIMITED_RESOURCES to the list here https://github.com/bluekitchen/btstack/blob/master/src/classic/hfp.c#L849
- try without ENABLE_HFP_HF_SAFE_SETTINGS. This was requested by a customer, but as far as I know, the stack is "more compatible" without it.

Cheers
  Matthias

amorniroli, 2025-04-09:
> Hi Alessandro.
>
> The log looks similar, good catch:
>
> Could you:
>
> * try adding ERROR_CODE_CONNECTION_REJECTED_DUE_TO_LIMITED_RESOURCES to the list here [...]

Hi @mringwal 

thanks for the hint.

- tried without ENABLE_HFP_HF_SAFE_SETTINGS the issue persists;
- tried adding ERROR_CODE_CONNECTION_REJECTED_DUE_TO_LIMITED_RESOURCES  to the list and it's working as expected.

It's accepting S1 as settings (S2 was refused due to ERROR_CODE_CONNECTION_REJECTED_DUE_TO_LIMITED_RESOURCES).

Perfect!

closed 2025-04-10 by mringwal