bluekitchen/btstack #676: hog_mouse_demo connection failing with Windows

peterharperuk, 2025-04-11:
**Describe the bug**

I'm testing a change to the pico-sdk on Pico 2 W. I'm using hog_mouse_demo. I connect to the "hid device" from Windows and Windows always seems to think it's connected. However, if I remove the device and repeat the pairing, the example doesn't seem to work every other time I do this. The example gives me an error "Mouse not connected, ignoring 'a'" even though Windows is not connected.

The example gets the connection handle from a HCI_EVENT_HIDS_META event. However this doesn't seem to get received in the failing cases.

Version: 7.95.61 (abcd531 CY) CRC: 4528a809 Date: Wed 2023-01-11 10:29:38 PST Ucode Ver: 1043.2169 FWID 01-7afb0879
cyw43 loaded ok, mac 2c:cf:67:be:08:04
BT FW download, version = CYW4343A2_001.003.016.0031.0000_Generic_SDIO_37MHz_wlbga_BU_dl_signed
BTstack up and running on 2C:CF:67:BE:08:05.
LE Connection Complete: <-- ADDED THE DEVICE IN WINDOWS
- Connection Interval: 60.00 ms
- Connection Latency: 0
Just Works requested
LE Connection Update:
- Connection Interval: 15.00 ms
- Connection Latency: 0
HCI_EVENT_HIDS_META <-- GOT THE EVENT
Report Characteristic Subscribed 1
LE Connection Update:
- Connection Interval: 15.00 ms
- Connection Latency: 0
HCI_EVENT_HIDS_META
Mouse: -30/0 - buttons: 00 <-- EVERYTHING WORKING
Disconnected <-- REMOVED THE DEVICE IN WINDOWS
LE Connection Complete: <-- ADDED THE DEVICE AGAIN
- Connection Interval: 60.00 ms
- Connection Latency: 0
Just Works requested
LE Connection Update:
- Connection Interval: 15.00 ms
- Connection Latency: 0
LE Connection Update:
- Connection Interval: 15.00 ms
- Connection Latency: 0
Mouse not connected, ignoring 'a'  <-- MISSING HCI_EVENT_HIDS_META 

A possible fix which works for me is to get the connection handle from the HCI_EVENT_META_GAP event? I can raise a PR for this if it sounds sensible.

**To Reproduce**

Steps to reproduce the behavior:
1. Run hog_mouse_demo on Windows 11
2. Add the device from Windows and it should connect.
3. Verify that the Pico device thinks it is connected by moving the mouse pointer (e.g. press A key in the console)
4. Remove the device from Windows
5. Repeat the process. It will only work every other time.

**Expected behavior**

The example should work every time?

**HCI Packet Logs**
For all Bluetooth issues, please provide a full packet log that includes initial pairing. You need to zip the packet log as GitHub does not know how to handle .pklg files. 

Will attach the log.

**Environment:  (please complete the following information):**
 - develop version of pico-sdk, btstack is at this commit. I'm using an oldish Windows 11 laptop.
commit 501e6d2b86e6c92bfb9c390bcf55709938e25ac1
Author: Matthias Ringwald <matthias@ringwald.ch>
Date:   Mon Jan 13 18:51:04 2025 +0100

    gap: store link key for standard/non-SSP pairing

**Additional context**
I haven't checked if this is a regression.

peterharperuk, 2025-04-11:
[minicom_btstack_11apr.zip](https://github.com/user-attachments/files/19706017/minicom_btstack_11apr.zip)

peterharperuk, 2025-04-11:
This is probably something funny about CYW43. If I power cycle the device the connections work ok.

mringwal, 2025-04-14:
Quick question: it looks like you're deleting the bonding on windows after each pairing & mousing.

Does reconnect work if you only disconnect it or power cycle the Pico?
In general, the bonding & pairing should work every time.

peterharperuk, 2025-04-14:
> it looks like you're deleting the bonding on windows after each pairing & mousing.

Yes, I was deleting the device in Windows and trying to pair again.

> Does reconnect work if you only disconnect it or power cycle the Pico?

Yes if it paired properly and the got the connection handle then it will also work after a power cycle. But if it doesn't get HCI_EVENT_HIDS_META event then a power cycle doesn't help, it behaves the same - no connection handle.

mringwal, 2025-04-21:
Hi Peter. I've just tried to connect to hog_mouse_demo on libusb port from my iPhone. I cannot pair from the Settings dialog, but any GATT Browser like LIghtBlue or nRF Toolkit is enough to trigger the connection.

After pairing, I've removed the bonding in the Settings and re-connected. This works correctly every time.

Could you come closer to my environment?
- use btstack/develop version
- pair/test from iOS or Android
 
In your log, both connection attempts are similar. connect, gatt query, pairing.
However, only in the first attempt does the remote device (Windows) subscribe to the Report Characteristic - which is the one to receive key presses or mouse movements. BTstack is required to remember which notifications are enabled - but we clear all stored information if there's a new pairing. If you delete all bonding information on Windows, I don't think it should assume that we remembered that subscription. It's also curious why it works every other time.

Could you try with some other device: iOS, Android, Mac? :)

mringwal, 2025-06-12:
No response -> cannot help. Feel free to open a new issue when needed.

closed 2025-06-12 by mringwal