bluekitchen/btstack #679: HOG host not working with ble mice

baralgin1003, 2025-04-21:
**Describe the bug**

I have several cheap BLE mice, defined as "BT5.2 Mouse". I run example hog_host_demo on esp32 Wroom. Both my mice are found, but do not pair. At the same time, the Logi MX Master 3S works well. The mice were bought in different places and have different designs, but most likely they have the same controller. These are quite popular models of "dual-mode" mice (2.4G + BT) and I think there are dozens of other models with the same hardware. I would really like to make them work.
Thank you!

**HCI Packet Logs**
Mouse 1:
Found, connect to device with public address 13:08:AC:00:0E:14 ...
[00:01:21.773] LOG -- hci.c.328: create_connection_for_addr 13:08:AC:00:0E:14, type 0
[00:01:21.773] LOG -- hci.c.8683: gap_connect: send create connection next
[00:01:21.783] EVT <= 3E 2B 02 01 00 00 14 0E 00 AC 08 13 1F 02 01 05 03 03 12 18 03 19 C2 03 06 FF 06 00 03 00 80 0C 09 42 54 35 2E 32 20 4D 6F 75 73 65 C2 
[00:01:21.793] EVT <= 3E 2B 02 01 03 01 37 24 33 BA 7D 32 1F 1E FF 06 00 01 09 20 22 B1 94 94 AD B2 D0 DF B0 5D DC 10 B2 CC 69 7C 11 B0 C4 69 E4 EE 1F A8 AC 
[00:01:21.813] EVT <= 0E 04 05 0C 20 00
[00:01:21.813] LOG -- hci.c.7360: sending hci_le_create_connection
[00:01:21.823] CMD => 0D 20 19 60 00 30 00 00 00 14 0E 00 AC 08 13 00 08 00 18 00 04 00 48 00 00 00 00 00
[00:01:21.823] EVT <= 0F 04 00 05 0D 20
Timeout - abort connection
[00:01:31.773] CMD => 0E 20 00 
Scanning for LE HID devices...
[00:01:31.773] EVT <= 0E 04 05 0E 20 00 
[00:01:31.783] EVT <= 3E 13 01 02 FF FF 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

Mouse 2:
Found, connect to device with public address C3:20:24:01:5E:B9 ...
[00:00:45.523] LOG -- hci.c.328: create_connection_for_addr C3:20:24:01:5E:B9, type 0
[00:00:45.523] LOG -- hci.c.8683: gap_connect: send create connection next
[00:00:45.533] EVT <= 3E 2B 02 01 00 00 B9 5E 01 24 20 C3 1F 02 01 05 03 03 12 18 03 19 C2 03 06 FF 06 00 03 00 80 0C 09 42 54 35 2E 32 20 4D 6F 75 73 65 C8 
[00:00:45.543] EVT <= 0E 04 05 0C 20 00
[00:00:45.553] LOG -- hci.c.7360: sending hci_le_create_connection
[00:00:45.553] CMD => 0D 20 19 60 00 30 00 00 00 B9 5E 01 24 20 C3 00 08 00 18 00 04 00 48 00 00 00 00 00
[00:00:45.563] EVT <= 0F 04 00 05 0D 20
Timeout - abort connection
[00:00:55.533] CMD => 0E 20 00 
Scanning for LE HID devices...

**Environment:  (please complete the following information):**
 - Current BTstack branch: develop
 - Bluetooth Controller ESP32 WROOM
 - Remote device: "BT5.2 Mouse"

**Additional context**
Add any other context about the problem here.

mringwal, 2025-04-22:
It's not clear if these mice support Bluetooth Classic HID or Bluetooth LE HOG.
Could you try to connect using hid_host_demo?

If that fails, to please post the full log of both tools after converting it to .pklg with tool/create_packet_log.py.

baralgin1003, 2025-04-22:
these mice only support BLE connection. i updated the log of the second mouse, please check, maybe there will be more useful information in it

mringwal, 2025-04-23:
Where's the log? Please capture the full log after restart into a file, try to connect and then converting it to .pklg with tool/create_packet_log.py.

mringwal, 2025-04-28:
> baralgin1003
>  left a comment
> (bluekitchen/btstack#679)
> Hello! I collected logs (I hope correctly, rename to *.pklg) from example hog_host_demo. I checked the mouse via NRF Connect, it really has only BLE connection. If you have any questions, please let me know. log-mouse52.pklg.txt <https://github.com/user-attachments/files/19929301/log-mouse52.pklg.txt>

Hi. In the log, the mouse responds with an error to BTstack’s Exchange MTU request. 

Could you try to call 
gatt_client_mtu_enable_auto_negotiation(0) after gatt_client_init() and retry / send a new log?

baralgin1003, 2025-04-28:
done!
[log-mouse52.pklg.txt](https://github.com/user-attachments/files/19943457/log-mouse52.pklg.txt)

mringwal, 2025-04-28:
The log does not contain a connection. Fell free to open the .pklg file with Wireshark to check :)

baralgin1003, 2025-04-28:
It seems... it worked? Зlease take a look. Should I always include gatt_client_mtu_enable_auto_negotiation(0) or make it optional? [log-mouse52.pklg.txt](https://github.com/user-attachments/files/19943713/log-mouse52.pklg.txt)

baralgin1003, 2025-04-30:
Thank you! It works!

mringwal, 2025-04-30:
Awesome - thanks for exposing this corner case!

closed 2025-04-30 by mringwal