bluekitchen/btstack #684: Browsing Example, ESP32 Reboot on Audio Play

thejacko12354, 2025-05-05:
**Describe the bug**

If you start audio from a Device (iOS or Android) on the avrcp_browsing example the esp32 crashes.

**To Reproduce**

Steps to reproduce the behavior:
1. Run example 'avrcp_browsing'
2. Connect to esp, from iOS or Android
3. Start play in Apple Music

**Expected behavior**

Don't crash :D

**HCI Packet Logs**
[https://www.icloud.com/iclouddrive/076SlIH1N346UxfaRbD8iPpvQ#log](url)

**Environment:  (please complete the following information):**
 - Current BTstack branch: master
 - Bluetooth Controller: ESP32
 - Remote device: iPhone 13 Pro iOS 18.4

**Additional Log Infos**
`assert failed: avdtp_packet_handler avdtp.c:887 (avdtp_sink_handle_media_data)

Backtrace: 0x40081d1d:0x3ffbb8e0 0x40090189:0x3ffbb900 0x400967b1:0x3ffbb920 0x400ecfb2:0x3ffbba40 0x40128272:0x3ffbba90 0x400e9b8e:0x3ffbbab0 0x400eb41a:0x3ffbbaf0 0x400eb995:0x3ffbbb10 0x400e1ae5:0x3ffbbb30 0x400e6ea5:0x3ffbbb50 0x400e703e:0x3ffbbb90 0x400ec088:0x3ffbbbb0 0x400f82f9:0x3ffbbbe0 0x400e1503:0x3ffbbc10 0x400dae40:0x3ffbbc30 0x4012a63f:0x3ffbbc50 0x40090d51:0x3ffbbc80
--- 0x40081d1d: panic_abort at C:/Users/Julian/esp/v5.4.1/esp-idf/components/esp_system/panic.c:454
--- 0x40090189: esp_system_abort at C:/Users/Julian/esp/v5.4.1/esp-idf/components/esp_system/port/esp_system_chip.c:87
--- 0x400967b1: __assert_func at C:/Users/Julian/esp/v5.4.1/esp-idf/components/newlib/assert.c:80
--- 0x400ecfb2: avdtp_packet_handler at C:/Users/Julian/esp/v5.4.1/esp-idf/components/btstack/src/classic/avdtp.c:887 (discriminator 1)
--- 0x40128272: l2cap_dispatch_to_channel at C:/Users/Julian/esp/v5.4.1/esp-idf/components/btstack/src/l2cap.c:1125
--- 0x400e9b8e: l2cap_acl_classic_handler_for_channel at C:/Users/Julian/esp/v5.4.1/esp-idf/components/btstack/src/l2cap.c:4808
--- 0x400eb41a: l2cap_acl_classic_handler at C:/Users/Julian/esp/v5.4.1/esp-idf/components/btstack/src/l2cap.c:4870
--- 0x400eb995: l2cap_acl_handler at C:/Users/Julian/esp/v5.4.1/esp-idf/components/btstack/src/l2cap.c:4949
--- 0x400e1ae5: hci_emit_acl_packet at C:/Users/Julian/esp/v5.4.1/esp-idf/components/btstack/src/hci.c:8043
--- 0x400e6ea5: acl_handler at C:/Users/Julian/esp/v5.4.1/esp-idf/components/btstack/src/hci.c:1248
--- 0x400e703e: packet_handler at C:/Users/Julian/esp/v5.4.1/esp-idf/components/btstack/src/hci.c:4816
--- 0x400ec088: transport_deliver_packets at C:/Users/Julian/esp/v5.4.1/esp-idf/components/btstack/btstack_port_esp32.c:200
--- 0x400f82f9: btstack_run_loop_freertos_execute at C:/Users/Julian/esp/v5.4.1/esp-idf/components/btstack/platform/freertos/btstack_run_loop_freertos.c:172
--- 0x400e1503: btstack_run_loop_execute at C:/Users/Julian/esp/v5.4.1/esp-idf/components/btstack/src/btstack_run_loop.c:310
--- 0x400dae40: app_main at C:/Users/Julian/Documents/idf-projects/btstack_avrcp/main/main.c:82
--- 0x4012a63f: main_task at C:/Users/Julian/esp/v5.4.1/esp-idf/components/freertos/app_startup.c:208
--- 0x40090d51: vPortTaskWrapper at C:/Users/Julian/esp/v5.4.1/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:139`

mringwal, 2025-05-05:
Hi. The assert is triggered because the AVRCP Browsing Client does not register a media handler.
We've updated the code on the develop branch to just ignore the audio data in this case.

Do you plan on building a jukebox?

Cheers
 Matthias

closed 2025-05-05 by mringwal

thejacko12354, 2025-05-05:
Hi Matthias,
ok thanks for the response. I compared the sink example and the browsing example and found the part with the handler.

Almost :) I'm currently building a BT Sink/AVRCP-Browsing -> iAP converter.

Cheers
Julian