bluekitchen/btstack #686: RPI Pico cortex  m33 if you enable unaligned usage fault trap btstack in hci.c gives a usage fault

tbandtg, 2025-05-22:
**Describe the bug**

Happens every time if you enable usage faults SCB_CCR_UNALIGN_TRP_Msk

**HCI Packet Logs**
#ifdef ENABLE_BLE
    printf("hci stack address %X rand address address %X", hci_stack,  hci_stack->le_random_address);
    for (int i = 0; i < 6; ++i) hci_stack->le_random_address[i] = 0;
    memset(hci_stack->le_random_address, 0, 6);
    hci_stack->le_random_address_set = 0;
#endif

mringwal, 2025-05-22:
In which line does the unaligned access happen exactly?
Yes, BTstack does not contain a single pack attribute as far as I know.

mringwal, 2025-05-22:
Does it only happen if you enable the SCB_CCR_UNALIGN_TRP_Msk?
If yes, why do you enable it? :)

tbandtg, 2025-05-22:
I mean yes, that is the very short term explanation, but I do not like leaving traps like that un enabled.

mringwal, 2025-05-23:
hci_stack->le_random_address has type bd_addr_t, which we defined as uint8_t [6], an array of 6 uint8_t.
As it's a uint8_t array, there's no reason for C to align it's start to a word (32-bit) boundary. In your test it's at 200045E6, which is not word / 4-byte aligned. It looks like the trap is triggered by the memset function.

Could you try to see if this also causes the trap?

uint8_t buffer[10];
memset(&buffer[1], 0, 4);
memset(&buffer[2], 0, 4);
 
If yes, then memset on the pico sdk does not bother to work with unaligned access. Please ask the experts there then.

tbandtg, 2025-05-27:
You called it, def. in the std library as a memcpy does the same. It is like the compiler doesn't realize that it is a size 1 and is copying a size 4 into it. seems odd. Ill post in their forums

mringwal, 2025-05-28:
Well. I don't consider an unaligned access a security risk. The Cortex-M33 supports unaligned access like for example x86. memset tries to be as fast as possible and will use 32-bit writes to speed up. Given that M33 supports unaligned access, there's no reason it should care for word alignment. If you want to avoid unaligned access, I guess you can compile libc yourself and use some flags to use an alternative implementation.

closed 2025-05-28 by mringwal