bluekitchen/btstack #687: WIFI and BLE cannot coexist when using RTL8723DU

shiyu-sz, 2025-05-24:
I use btstack on a linux platform (ubuntu24 and buildroot2022), using the RTL8723DU chip, and I found that if I don't load the WIFI driver, I run the gatt_counter examples normally. But if I load the WIFI driver, I get an error "hci_transport_h2_libusb.c.851: Error libusb_set_configuration: -6" when I run gatt_counter, which means "LIBUSB_ERROR_BUSY". Can't WIFI and BLE coexist?

mringwal, 2025-05-25:
I have no experience with Wifi but can imagine that additional rules apply. You may try our newish "linux" port in port/linux on develop and let the kernel load the wifi + Bluetooth firmware. Please let me know if/how that works for you.

shiyu-sz, 2025-05-26:
port/linux I tested on ubuntu24 and it was normal, BLE can be shared with WIFI, I moved to buildroot2022 to try it