bluekitchen/btstack #688: RFCOMM credits stop to be received

amorniroli, 2025-05-27:
## Bug Description

The SPP Streamer demo ceases data transmission unexpectedly. This behavior appears to be due to the controller no longer sending ACL packets that include RFCOMM credit information. While this may be a controller-side issue, I would appreciate feedback from the BTstack perspective.

## Steps to Reproduce

1. Run the `spp_streamer.c` example.
2. Connect to the SPP server from a remote device (in this case, a Windows 10 system using a virtual Bluetooth COM port, post-pairing).
3. Allow the streamer to run continuously without interaction.

## Expected Behavior

The streamer should continuously send data indefinitely without interruption.

## Observed Behavior

At some point during execution, ACL packets containing RFCOMM "new credits" are no longer received from the controller. As a result, the streamer halts transmission.

At 11.00:26 it's me sending a char via Teraterm for checking the communication.

[image]

## HCI Packet Logs

[wireshark.txt](https://github.com/user-attachments/files/20451741/wireshark.txt)

## Environment

- **BTstack Branch**: `develop` (commit `84b466728b02e02cc2e2aeb28408bfda98ac4707`)
- **Bluetooth Controller**: CYBT-343026-01 (CYW20706)
- **Remote Device**: Windows 10 with Tera Term terminal application

mringwal, 2025-05-29:
Hi @amorniroli The RFCOMM credits are managed by the RFCOMM protocol implementation in the Host Stack, so if it's not working correctly either one of the Host Stacks makes a mistake or a packet with Credits gets lost over the air (rather unlikely).

I had a look at the log and it seems like remote (the soon to be unsupported Windows 10 ;) is not providing new credits.
(BTstack logs the number of outgoing credits)

There are a few RFCOMM Remote Port Negotiation Commands/Responses which are used to configure the remote port (baudrate, data bits, parity, flowcontrol, ...). These are not common/needed for an active SPP connection. I don't think they are relevant to the credit-based flow control.

Could it be that Tera term is busy which cause the flow control to pause?

Could you:
a) try to run BTstack's SPP Streamer Client with an USB dongle on Windows for comparison?
b) try to use RFCOMM on Windows directly without the COM abstraction (which I find clumsy on any platform)?

amorniroli, 2025-06-06:
> Thank you for the feedback [@mringwal](https://github.com/mringwal) .
> 
> I'll try next days and let you know!

Sorry didn't have last days. I'll try next week, but it seems to be more a windows COM abstraction problem. No issues on other devices

mringwal, 2025-06-25:
Ok. I'm closing this now. Feel free to re-open if you suspect an issue with the BTstack side.

closed 2025-06-25 by mringwal